NatLabRockies/ROSCO #306: Flang support

marchdf, 2024-01-27:
Hi, I was using the flang compiler on Frontier today (which I was using for openfast) but I got errors in ROSCO when using the same compiler. I noticed in the cmake that ROSCO only supports GNU and Intel compilers: https://github.com/NREL/ROSCO/blob/b994e4de67c423ff598a4bb354657b62ff61a50d/rosco/controller/CMakeLists.txt#L44. I was hoping there was a path towards supporting flang so that I don't have to mix compilers. Thanks!

marchdf, 2024-01-30:
Thanks for answering so quickly! If I get a chance I might give your suggestion a shot.

marchdf, 2024-02-05:
@dzalkind I needed this for frontier and I have a working version. But it's for ROSCO 2.7 because those are the input files I was given. Here's the branch: https://github.com/marchdf/ROSCO/tree/rosco-27-flang. Let me know if that "looks" ok and I can try doing the same for the dev version of ROSCO and submit a PR.

dzalkind, 2024-02-06:
Thanks, Marc!  It looks okay to me. Ideally, we'd figure out how to keep[ these lines](https://github.com/marchdf/ROSCO/commit/cb5005e08322def0aae5de18f0bfc7f6ca86b002#r138325127) as they are ROSCO's main debugging output.

Apparently there are a bunch of Flang flavors out there, with a new one due in March.  Maybe it will resolve the issue you raised.  Which Flang compiler you are using on frontier?

marchdf, 2024-02-06:
Yes I definitely think the eventual PR should have those lines in there. I am going to work with someone who knows fortran/flang better than me and see if we can resolve this. I am a bit confused by the flang landscape. Let me get back to on what kind of version I am using.

dzalkind, 2024-12-06:
@ndevelder is also interested in flang support (#397)

I don't have a set up with flang, so it's hard for me to make the required fixes to this source code.  If either of you are able to help me with that, I could support this better.

marchdf, 2024-12-06:
That would be fantastic and if you had something, I would be happy to test. Here is a link to my branch that made v2.7 work: https://github.com/marchdf/ROSCO/tree/rosco-27-flang. Please feel free to adapt for newer versions and I can try it out.

dzalkind, 2024-12-06:
Great! I'll bring your branch up to date.

The only real hang up was that flang didn't like this: https://github.com/marchdf/ROSCO/commit/cb5005e08322def0aae5de18f0bfc7f6ca86b002#diff-24535ca502db14768af01d975fd720e3fd71b2986252bd59cda15e89af2f13f2L519

That bit of code is used to see the internal signals of ROSCO, so it could be something we just omit from flang builds with a compile-time flag.

closed 2025-01-08 by marchdf