NatLabRockies/ROSCO #308: Running problem using FAST.Farm with IEA 15MW

domenicoruotolo, 2024-02-02:
Hello,

I'm trying to use FAST.Farm for a three IEA 15MW turbines in row layout but I'm having problem with the ServoDyn module, in particular with the ROSCO controller.
I'm using FAST.Farm-v3.5.1 and ROSCO-v2.9.0

This is the error I have:

`------------------------------------------------------------------------------
Running ROSCO-v2.9.0
A wind turbine controller framework for public use in the scientific field
Developed in collaboration: National Renewable Energy Laboratory
                            Delft University of Technology, The Netherlands
------------------------------------------------------------------------------
At line 773 of file /home/conda/feedstock_root/build_artifacts/rosco_1705704127312/work/rosco/controller/src/ReadSetParameters.f90 (unit = 11, file = '/home/domenico_ruotolo/iea_15MW/IEA-15-240-RWT/Cp_Ct_Cq.IEA15MW.txt')
Fortran runtime error: Bad real number in item 73 of list input

Error termination. Backtrace:
#0  0x7f6ef97914d3 in read_real
        at ../../../libgfortran/io/list_read.c:2025
#1  0x7f6ef9791c46 in list_formatted_read_scalar
        at ../../../libgfortran/io/list_read.c:2179
#2  0x7f6e5af4bff6 in ???
#3  0x7f6e5af51fbf in ???
#4  0x7f6e5af3fc1b in ???
#5  0x55de6ae5c8a6 in ???
#6  0x55de6ae5ca18 in ???
#7  0x55de6ae5ccc5 in ???
#8  0x55de6ae1c5e2 in ???
#9  0x55de6ae4151d in ???
#10  0x55de6a8d6f6e in ???
#11  0x55de6a8d7363 in ???
#12  0x55de6a84ff32 in ???
#13  0x55de6a850349 in ???
#14  0x55de6a729809 in ???
#15  0x55de6a725258 in ???
#16  0x55de6a78ea13 in ???
#17  0x55de6a5e0120 in ???
#18  0x7f6ef9307082 in ???
#19  0x55de6a5e01ac in ???`

I'm using the ROSCO controller IEA-15-240-RWT-Monopile_DISCON.IN (https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT-Monopile/IEA-15-240-RWT-Monopile_DISCON.IN). The DLL_Infile is Cp_Ct_Cq.IEA15MW.txt (https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT/Cp_Ct_Cq.IEA15MW.txt).
I also tried again after updating the DISCON.IN file to the newest version of ROSCO (https://github.com/NREL/ROSCO/blob/main/Examples/19_update_discon_version.py). But I have the same error.

Could you please help me solving this issue? Thank you.

Best regards,
Domenico Ruotolo

domenicoruotolo, 2024-02-06:
Hi!
The DLL _Infile is correctly pointing to the DISCON.IN file.
For example, this is the ServoDyn file for one of the three turbines. The DLL_Infile and DLL_FileName are in the right paths. 

`------- SERVODYN v1.05.* INPUT FILE --------------------------------------------
IEA 15 MW offshore reference model monopile configuration
---------------------- SIMULATION CONTROL --------------------------------------
False                  Echo        - Echo input data to <RootName>.ech (flag)
"default"              DT          - Communication interval for controllers (s) (or "default")
---------------------- PITCH CONTROL -------------------------------------------
5                      PCMode      - Pitch control mode {0: none, 3: user-defined from routine PitchCntrl, 4: user-defined from Simulink/Labview, 5: user-defined from Bladed-style DLL} (switch)
0.0                    TPCOn       - Time to enable active pitch control (s) [unused when PCMode=0]
9999.9                 TPitManS(1) - Time to start override pitch maneuver for blade 1 and end standard pitch control (s)
9999.9                 TPitManS(2) - Time to start override pitch maneuver for blade 2 and end standard pitch control (s)
9999.9                 TPitManS(3) - Time to start override pitch maneuver for blade 3 and end standard pitch control (s) [unused for 2 blades]
2.0                    PitManRat(1) - Pitch rate at which override pitch maneuver heads toward final pitch angle for blade 1 (deg/s)
2.0                    PitManRat(2) - Pitch rate at which override pitch maneuver heads toward final pitch angle for blade 2 (deg/s)
2.0                    PitManRat(3) - Pitch rate at which override pitch maneuver heads toward final pitch angle for blade 3 (deg/s) [unused for 2 blades]
0.0                    BlPitchF(1) - Blade 1 final pitch for pitch maneuvers (degrees)
0.0                    BlPitchF(2) - Blade 2 final pitch for pitch maneuvers (degrees)
0.0                    BlPitchF(3) - Blade 3 final pitch for pitch maneuvers (degrees) [unused for 2 blades]
---------------------- GENERATOR AND TORQUE CONTROL ----------------------------
5                      VSContrl    - Variable-speed control mode {0: none, 1: simple VS, 3: user-defined from routine UserVSCont, 4: user-defined from Simulink/Labview, 5: user-defined from Bladed-style DLL} (switch)
1                      GenModel    - Generator model {1: simple, 2: Thevenin, 3: user-defined from routine UserGen} (switch) [used only when VSContrl=0]
95.756                 GenEff      - Generator efficiency [ignored by the Thevenin and user-defined generator models] (%)
True                   GenTiStr    - Method to start the generator {T: timed using TimGenOn, F: generator speed using SpdGenOn} (flag)
True                   GenTiStp    - Method to stop the generator {T: timed using TimGenOf, F: when generator power = 0} (flag)
9999.9                 SpdGenOn    - Generator speed to turn on the generator for a startup (HSS speed) (rpm) [used only when GenTiStr=False]
0.0                    TimGenOn    - Time to turn on the generator for a startup (s) [used only when GenTiStr=True]
9999.9                 TimGenOf    - Time to turn off the generator (s) [used only when GenTiStp=True]
---------------------- SIMPLE VARIABLE-SPEED TORQUE CONTROL --------------------
7.559987120819503      VS_RtGnSp   - Rated generator speed for simple variable-speed generator control (HSS side) (rpm) [used only when VSContrl=1]
19624046.66639         VS_RtTq     - Rated generator torque/constant generator torque in Region 3 for simple variable-speed generator control (HSS side) (N-m) [used only when VSContrl=1]
343357.4355671095      VS_Rgn2K    - Generator torque constant in Region 2 for simple variable-speed generator control (HSS side) (N-m/rpm^2) [used only when VSContrl=1]
2.                     VS_SlPc     - Rated generator slip percentage in Region 2 1/2 for simple variable-speed generator control (%) [used only when VSContrl=1]
---------------------- SIMPLE INDUCTION GENERATOR ------------------------------
9999.9                 SIG_SlPc    - Rated generator slip percentage (%) [used only when VSContrl=0 and GenModel=1]
9999.9                 SIG_SySp    - Synchronous (zero-torque) generator speed (rpm) [used only when VSContrl=0 and GenModel=1]
9999.9                 SIG_RtTq    - Rated torque (N-m) [used only when VSContrl=0 and GenModel=1]
9999.9                 SIG_PORt    - Pull-out ratio (Tpullout/Trated) (-) [used only when VSContrl=0 and GenModel=1]
---------------------- THEVENIN-EQUIVALENT INDUCTION GENERATOR -----------------
9999.9                 TEC_Freq    - Line frequency [50 or 60] (Hz) [used only when VSContrl=0 and GenModel=2]
100                    TEC_NPol    - Number of poles [even integer > 0] (-) [used only when VSContrl=0 and GenModel=2]
9999.9                 TEC_SRes    - Stator resistance (ohms) [used only when VSContrl=0 and GenModel=2]
9999.9                 TEC_RRes    - Rotor resistance (ohms) [used only when VSContrl=0 and GenModel=2]
9999.9                 TEC_VLL     - Line-to-line RMS voltage (volts) [used only when VSContrl=0 and GenModel=2]
9999.9                 TEC_SLR     - Stator leakage reactance (ohms) [used only when VSContrl=0 and GenModel=2]
9999.9                 TEC_RLR     - Rotor leakage reactance (ohms) [used only when VSContrl=0 and GenModel=2]
[... cut ...]
"ES10.3E2"             OutFmt      - Format used for text tabular output (except time).  Resulting field should be 10 characters. (quoted string) (currently unused)
0.0                    TStart      - Time to begin tabular output (s) (currently unused)
              OutList      - The next line(s) contains a list of output parameters.  See OutListParameters.xlsx for a listing of available output channels, (-)
"GenPwr"
"GenTq"
END of input file (the word "END" must appear in the first 3 columns of this last OutList line)
---------------------------------------------------------------------------------------
`

dzalkind, 2024-02-06:
Hi,

This error appears when there is a mismatch between the [size of the Cp table](https://github.com/NREL/ROSCO/blob/b994e4de67c423ff598a4bb354657b62ff61a50d/Examples/Test_Cases/IEA-15-240-RWT-UMaineSemi/Cp_Ct_Cq.IEA15MW.txt#L4) and the [sizes provided in the DISCON](https://github.com/NREL/ROSCO/blob/b994e4de67c423ff598a4bb354657b62ff61a50d/Examples/Test_Cases/IEA-15-240-RWT-UMaineSemi/DISCON-UMaineSemi.IN#L113).

I would recommend starting with [a working example, like this one](https://github.com/NREL/ROSCO/tree/main/Examples/Test_Cases/IEA-15-240-RWT-UMaineSemi).  You can disable the floating DOFs to make it more like your desired simulation.  Then make small changes to reach your desired simulation configuration.

On our end, we will try out your simulation set up and work towards improving the error messaging around this issue.

I hope this helps.

Best, Dan

dzalkind, 2024-02-13:
Has this issue been resolved?  Better error catching has been added in #314

domenicoruotolo, 2024-02-15:
Yes, thanks!

closed 2024-02-15 by dzalkind