NatLabRockies/ROSCO #313: ZeroMW client not properly installed | Example 17

Ian-Ammerman, 2024-02-09:
I have installed ROSCO 2.9 following the full installation instructions (2.1) here: https://rosco.readthedocs.io/en/latest/source/install.html#full-rosco

Within WSL I create the conda environment rosco-env with python 3.10 and activate it. I navigate to ROSCO and pip install -e . which succesfully builds ROSCO. I am then able to run example 01.

I then install OpenFAST via conda install -c conda-forge openfast. Running example 17a or 17b then yields the following error:

ROSCO:PitchControl:PitchSaturation:interp1d:interp1d:interp1d:interp1d:interp1d:VariableSpeedControl:WindSpeedEstimator:AeroDynTorque:interp2d:interp2d:interp1d:UpdateZeroMQ: >> The ZeroMQ client has not been properly installed, please install it to use ZMQ_Mode > 0.

Installing zeroMQ using anaconda (conda install zeromq & conda install pyzmq) has no effect. Is there another install procedure I should follow perhaps?

dzalkind, 2024-02-09:
[This is where cmake looks](https://github.com/NREL/ROSCO/blob/b994e4de67c423ff598a4bb354657b62ff61a50d/rosco/controller/CMakeLists.txt#L67) for the `libzmq` library.  You can search for it with `ldd libzmq` I think.

You'll also want to delete the `build/` directories in the ROSCO main folder and `ROSCO/rosco/controller` before re-trying the install with `pip install -e .`

If all else fails, you can install `libzmq` with `apt-get install libzmq3-dev`.

I hope this helps.

Best, Dan

Ian-Ammerman, 2024-02-09:
Thanks for the response,

After cloning ROSCO from github there is no build/ directory in either the main ROSCO folder or controller subfolder when I run pip install -e . 

Running a search via ldd yields no such file or directory (run 1 level up from ROSCO main folder):

`(rosco-env) ian_ammerman@Dell-Laptop:~$ sudo apt-get install libzmq3-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libzmq3-dev is already the newest version (4.3.4-2).
0 upgraded, 0 newly installed, 0 to remove and 54 not upgraded.`

After which, I am still unable to locate the library with ldd or run the example:

`(rosco-env) ian_ammerman@Dell-Laptop:~$ ldd libzmq
ldd: ./libzmq: No such file or directory`

Note that the same result is observed with ldd libzmq3 and libzmq3-dev as well in the ldd command.

I notice in the FORTRAN you linked reference to PkgConfig. Earlier I was compiling just the controller following steps in 2.2.3 with CMake (I deleted everything and restarted before steps above).  The following error was given when running the `cmake ..` command

`Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)`

Might this be relevant?

gbarter, 2024-02-09:
If you are building within a conda environment, it is best to do: `conda install pkg-config`.

@dzalkind : If that is the issue. We should probably add `pkg-config` to the environment file.

Ian-Ammerman, 2024-02-09:
@gbarter Thank you for the input - I have had success by running `sudo apt-get pkg-config` after creating the environment but before running the ROSCO installation. I was then able to compile and run example 17a and 17b.

gbarter, 2024-02-09:
Glad to hear it!

abhineet-gupta, 2024-02-15:
FYI, for non linux systems using `conda install pkg-config` also solves this problem.

fredrikfselheim, 2024-02-16:
Hi. I'm experiencing the same issue as Ian did, the zeromq client not being properly installed. However, I'm working with MacOS arm64 architecture and are therefore not able to run the `sudo apt-get pkg-config`. Also the `pip install -e .` command does not run successfully on my system (wisdem, meson.build error) and I've only been able to compile the DLL through running `python setup.py develop`. Would be great to get some input on how to deal with this!

gbarter, 2024-02-16:
On a Mac, you would do `conda install zeromq pyzmq pkg-config`.  If you are trying to install system-wide, then you could do `brew install pkg-config` and `brew install zeromq` or an equivalent for Macports if that is your custom.

I am also confused by your meson error as ROSCO doesn't use meson.  Are you building WISDEM from source too?  If you are more interested in ROSCO, please do `conda install wisdem`.  If you are having trouble building WISDEM from source, please see the [WISDEM](https://github.com/WISDEM/WISDEM/) repo.

gbarter, 2024-02-16:
I think your conda environment is likely confused with having simultaneous `pip install` and `conda install` packages.  Taking a closer look at the installation instructions, I think this is our fault and not your fault.  Here is my recommendation on a fresh start that should give you a ROSCO package that is built from source on Mac ARM:

```
conda config --add channels conda-forge
conda env create --name rosco-env -f https://raw.githubusercontent.com/NREL/ROSCO/master/environment.yml python=3.11
conda activate rosco-env
conda install -y git compilers
git clone https://github.com/NREL/ROSCO.git
cd ROSCO
pip install -e . -v
```

@dzalkind - I think the [install instructions](https://rosco.readthedocs.io/en/latest/source/install.html) are a bit stale.  Can we add this to our to-do list?  Perhaps it would be best to add something to the README too?

fredrikfselheim, 2024-02-16:
Yes, thank you! I fresh start would definitely not hurt.

I get this when running your second command:
 rosco-env -f https://raw.githubusercontent.com/NREL/ROSCO/master/environment.yml python=3.11

SpecNotFound: Invalid name 'python=3.11', try the format: user/package

abhineet-gupta, 2024-02-16:
Hi Fredrik,
The following commands work on an m1 mac. I installed some common dependencies using conda to avoid any system level conflict. Could you try the following too?

```
conda create -n RoscoNewInstall python=3.10
conda activate RoscoNewInstall
conda install wisdem pyzmq pkg-config openfast compilers
git clone https://github.com/NREL/ROSCO.git
cd ROSCO/
pip install -e .
cd Examples/
python ./17b_zeromq_multi_openfast.py 
```

gbarter, 2024-02-16:
> Yes, thank you! I fresh start would definitely not hurt.
> 
> I get this when running your second command: rosco-env -f https://raw.githubusercontent.com/NREL/ROSCO/master/environment.yml python=3.11
> 
> SpecNotFound: Invalid name 'python=3.11', try the format: user/package

Sorry about that, just remove the `python=3.11` from the command

fredrikfselheim, 2024-02-16:
Hi Abhineet
Thank you for responding. I'm afraid this gave me the same error as earlier for `pip install -e .` See 
[TerminalLog.pdf](https://github.com/NREL/ROSCO/files/14313289/TerminalLog.pdf) for details.

fredrikfselheim, 2024-02-16:
> conda config --add channels conda-forge
> conda env create --name rosco-env -f https://raw.githubusercontent.com/NREL/ROSCO/master/environment.yml python=3.11
> conda activate rosco-env
> conda install -y git compilers
> git clone https://github.com/NREL/ROSCO.git
> cd ROSCO
> pip install -e . -v

Sorry to say that the error with meson setup command and requirements to build wheel fails again. Please check the
[TerminalLog2.pdf](https://github.com/NREL/ROSCO/files/14313360/TerminalLog2.pdf) for details if you'd like.

gbarter, 2024-02-16:
Can you confirm that you have activated the `rosco-env` conda environment?  It isn't installing WISDEM from conda and trying to do it from pip instead.

fredrikfselheim, 2024-02-16:
Yes I have `rosco-env` activated when running `pip install -e . -v`

dzalkind, 2024-02-16:
It looks like the installation process keeps trying to install wisdem from cache via pip

```
Using cached wisdem-3.13.0.tar.gz (5.4 MB)
Running command pip subprocess to install build dependencies
```

Maybe try a
`python -m pip cache purge`

Then `conda install wisdem`

Then reinstall rosco with `pip install -e .`

fredrikfselheim, 2024-02-16:
Hi Dan, thanks for responding!
It still gave me the error 
[TerminalLog3.pdf](https://github.com/NREL/ROSCO/files/14313583/TerminalLog3.pdf)
I did
`python` -m pip cache purge`
`conda install wisdem`
`pip install -e .`
and I tried
`python` -m pip cache purge`
`conda remove wisdem`
`conda install wisdem`
`pip install -e .`

gbarter, 2024-02-16:
It actually looks like `pip` isn't seeing any of the conda packages.  With wisdem now installed, try `python setup.py develop`

gbarter, 2024-02-16:
Another option would be `pip install --no-deps -e . -v`

gbarter, 2024-02-16:
> @gbarter, why does `wisdem` need to be [here](https://github.com/NREL/ROSCO/blob/3ebe819411d5fadf630287d20bf44ff572149c63/pyproject.toml#L53) and in the environment yaml?

The idea is to support both a pure-conda workflow and a pure-pip workflow.  The conda one is generally more robust and pip is supposed to respect conda packages (which isn't working here). Clearly, the wisdem pypi package for Mac ARM also isn't working. I'm confident that we can figure these problems out and specify a good recipe in the documentation.  More items for the to-do list for sure.

Another suggestion is to more prominently encourage simply `conda install rosco` as the only needed step and then work from there.  I think this is deep in the instructions currently.

dzalkind, 2024-02-16:
Some notes based on a debugging session with Frederik:

- `pip install -e .` was not installing the rosco module properly.  We were able to work around the wisdem issue by deleting that line from the `.toml` file.
- I think we should probably post the conda installation procedure, and add the necessary dependencies to the environment yaml
```
conda env create --name rosco-env -f https://raw.githubusercontent.com/NREL/ROSCO/master/environment.yml 
conda install wisdem pyzmq pkg-config openfast compilers
```
Seemed to get us there.  We stopped when Frederik had other path issues related to odd file structure.  I suggested re-cloning and following the above steps.

fredrikfselheim, 2024-02-16:
Hi guys,

I now finally got it right! Thank you all for helping.

This is the approach that worked for me, if perhaps you’d like to know.
First of all deleting the rosco folder and rosco-env to get a fresh start.

conda config --add channels conda-forge
conda env create --name rosco-env -f https://raw.githubusercontent.com/NREL/ROSCO/master/environment.yml
conda activate rosco-env
conda install -y git compilers
conda install wisdem zeromq pyzmq pkg-config openfast compilers
git clone https://github.com/NREL/ROSCO.git
cd ROSCO
                python
                import rosco.toolbox

deleting wisdem from dependencies in pyproject.toml
pip install -e .

Again, thank you guys!
Have a great weekend

Best,
Fredrik

Dato: fredag, 16. februar 2024 kl. 14:41
Til: NREL/ROSCO ***@***.***>
Kopi: fredrikfselheim ***@***.***>, Comment ***@***.***>

closed 2024-09-03 by dzalkind