NatLabRockies/ROSCO #322: Individual pitch control (IPC) tuning, example "12_tune_ipc.py"

TordMalmei, 2024-02-26:
Hi, 
I want to use the individual pitch controller in ROSCO with OpenFAST. I try to tune the individual pitch controller gains (Kp and Ki) for the IEA-15-240-RWT-UMaineSemi by the example "12_tune_ipc.py". I have the complete ROSCO installation in a new environment in Anaconda. The code runs but will never end and I can't figure out why. 

The code runs but doesn't get any further than this: 
[image]

Has anyone experienced this before and can help get the IPC tuned? Or can someone provide me withe the gains neded so I can manually change the DISCON.IN file?

[image]

Cheers Tord

dzalkind, 2024-02-27:
Hi Tord,

I have an unpublished IPC set up for the IEA-15MW on my local machine with:

IPC_KI = [8.e-11, 0]
IPC_SatMode = 2
IPC_IntSat = 0.3

I hope this helps.  It may not be the best configuration, but it's a starting point.

If you CTRL+C when the script runs, can you see where it is stuck?

I'll work on debugging on my end, since your script should run on Windows, but there have been some growing pains with the new set up.  Also, I think we are a little overdue for a semi-automated IPC tuning.   I'll leave this issue open for both of these.

Best, Dan

TordMalmei, 2024-02-28:
Hi Dan,
Thank you very much for the answer and the numbers.
I tried to CTRL+SHIFT+C and got the "code analysis" tab that said "Source code has not been rated yet". I didn't get any more information.

Tord

dzalkind, 2024-05-08:
Any updates here or can we close this issue?