NatLabRockies/ROSCO #327: Error while installing ROSCO complitly on Windows 11

AdrianEolion, 2024-03-21:
Hi,

I am trying to install ROSCO on my computer on Windows so I can use it with OpenFAST, but I am having trouble with the installation. I have done all the steps on the installation guide and this error happens on the last command.

pip install -e .

(rosco-env) C:\Users\adria\OpenFAST\ROSCO-v.2.9.1>pip install -e .
Defaulting to user installation because normal site-packages is not writeable
Obtaining file:///C:/Users/adria/OpenFAST/ROSCO-v.2.9.1
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Requirement already satisfied: control in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from rosco==2.9.0) (0.9.4)
Requirement already satisfied: numpy in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from rosco==2.9.0) (1.26.4)
Requirement already satisfied: matplotlib in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from rosco==2.9.0) (3.8.3)
Requirement already satisfied: scipy in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from rosco==2.9.0) (1.12.0)
Requirement already satisfied: pandas in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from rosco==2.9.0) (2.2.1)
Requirement already satisfied: pyparsing in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from rosco==2.9.0) (3.1.2)
Requirement already satisfied: pyYAML in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from rosco==2.9.0) (6.0.1)
Requirement already satisfied: pyzmq in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from rosco==2.9.0) (25.1.2)
Requirement already satisfied: treon in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from rosco==2.9.0) (0.1.4)
Requirement already satisfied: wisdem in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from rosco==2.9.0) (3.13.0)
Requirement already satisfied: ruamel.yaml in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from rosco==2.9.0) (0.18.6)
Requirement already satisfied: contourpy>=1.0.1 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from matplotlib->rosco==2.9.0) (1.2.0)
Requirement already satisfied: cycler>=0.10 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from matplotlib->rosco==2.9.0) (0.12.1)
Requirement already satisfied: fonttools>=4.22.0 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from matplotlib->rosco==2.9.0) (4.50.0)
Requirement already satisfied: kiwisolver>=1.3.1 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from matplotlib->rosco==2.9.0) (1.4.5)
Requirement already satisfied: packaging>=20.0 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from matplotlib->rosco==2.9.0) (24.0)
Requirement already satisfied: pillow>=8 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from matplotlib->rosco==2.9.0) (10.2.0)
Requirement already satisfied: python-dateutil>=2.7 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from matplotlib->rosco==2.9.0) (2.9.0.post0)
Requirement already satisfied: pytz>=2020.1 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from pandas->rosco==2.9.0) (2024.1)
Requirement already satisfied: tzdata>=2022.7 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from pandas->rosco==2.9.0) (2024.1)
Requirement already satisfied: ruamel.yaml.clib>=0.2.7 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from ruamel.yaml->rosco==2.9.0) (0.2.8)
Requirement already satisfied: nbconvert in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from treon->rosco==2.9.0) (7.16.2)
Requirement already satisfied: jupyter-client in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from treon->rosco==2.9.0) (8.6.1)
Requirement already satisfied: jupyter in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from treon->rosco==2.9.0) (1.0.0)
Requirement already satisfied: docopt in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from treon->rosco==2.9.0) (0.6.2)
Requirement already satisfied: dearpygui in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from wisdem->rosco==2.9.0) (1.11.1)
Requirement already satisfied: jsonschema in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from wisdem->rosco==2.9.0) (4.21.1)
Requirement already satisfied: marmot-agents>=0.2.5 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from wisdem->rosco==2.9.0) (0.2.5)
Requirement already satisfied: moorpy in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from wisdem->rosco==2.9.0) (1.0.2)
Requirement already satisfied: nlopt in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from wisdem->rosco==2.9.0) (2.7.1)
Requirement already satisfied: openmdao<3.28 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from wisdem->rosco==2.9.0) (3.27.0)
Requirement already satisfied: openpyxl in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from wisdem->rosco==2.9.0) (3.1.2)
Requirement already satisfied: pydoe2 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from wisdem->rosco==2.9.0) (1.3.0)
Requirement already satisfied: python-benedict in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from wisdem->rosco==2.9.0) (0.33.2)
Requirement already satisfied: simpy in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from wisdem->rosco==2.9.0) (4.1.1)
Requirement already satisfied: sortedcontainers in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from wisdem->rosco==2.9.0) (2.4.0)
Requirement already satisfied: statsmodels in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from wisdem->rosco==2.9.0) (0.14.1)
Requirement already satisfied: networkx>=2.0 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from openmdao<3.28->wisdem->rosco==2.9.0) (3.2.1)
Requirement already satisfied: requests in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from openmdao<3.28->wisdem->rosco==2.9.0) (2.31.0)
Requirement already satisfied: six>=1.5 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from python-dateutil>=2.7->matplotlib->rosco==2.9.0) (1.16.0)
Requirement already satisfied: attrs>=22.2.0 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from jsonschema->wisdem->rosco==2.9.0) (23.2.0)
Requirement already satisfied: jsonschema-specifications>=2023.03.6 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from jsonschema->wisdem->rosco==2.9.0) (2023.12.1)
Requirement already satisfied: referencing>=0.28.4 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from jsonschema->wisdem->rosco==2.9.0) (0.34.0)
Requirement already satisfied: rpds-py>=0.7.1 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from jsonschema->wisdem->rosco==2.9.0) (0.18.0)
Requirement already satisfied: notebook in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from jupyter->treon->rosco==2.9.0) (7.1.2)
Requirement already satisfied: qtconsole in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from jupyter->treon->rosco==2.9.0) (5.5.1)
Requirement already satisfied: jupyter-console in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from jupyter->treon->rosco==2.9.0) (6.6.3)
Requirement already satisfied: ipykernel in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from jupyter->treon->rosco==2.9.0) (6.29.3)
Requirement already satisfied: ipywidgets in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from jupyter->treon->rosco==2.9.0) (8.1.2)
Requirement already satisfied: jupyter-core!=5.0.*,>=4.12 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from jupyter-client->treon->rosco==2.9.0) (5.7.2)
Requirement already satisfied: tornado>=6.2 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from jupyter-client->treon->rosco==2.9.0) (6.4)
Requirement already satisfied: traitlets>=5.3 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from jupyter-client->treon->rosco==2.9.0) (5.14.2)
Requirement already satisfied: beautifulsoup4 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from nbconvert->treon->rosco==2.9.0) (4.12.3)
Requirement already satisfied: bleach!=5.0.0 in c:\programdata\anaconda3\envs\rosco-env\lib\site-packages (from nbconvert->treon->rosco==2.9.0) (6.1.0)
[... cut ...] BUILD_TYPE=Release', '-DPython3_ROOT_DIR=C:\\ProgramData\\anaconda3\\envs\\rosco-env', '-DCALL_FROM_SETUP_PY:BOOL=ON', '-DBUILD_SHARED_LIBS=ON', '-DCMAKE_Fortran_FLAGS=-ffree-line-length-0']' returned non-zero exit status 1.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building editable for rosco
Failed to build rosco
ERROR: Could not build wheels for rosco, which is required to install pyproject.toml-based projects

closed 2024-04-11 by AdrianEolion

DavidChen0429, 2024-04-15:
Hi Adrian, I encountered the same problem now, how did you solve it?

AdrianEolion, 2024-04-17:
If I am honest I dont fully know. I deleted the enviroment and reinstalled everything. It did work after that. Also see #332 if the examples do not work (only happens to me on Windows).

Maybe try to use python3 instead of python, or vice versa.

dzalkind, 2024-04-17:
You may need to install a few packages.  Try `conda install m2w64-toolchain libpython` and reinstalling.

giannikosk, 2024-04-17:
Hello, I am no master of the software, nor have I a great programming background, but for the last day I've been trying to get a hang of rosco and wisdem and encountered the same issue. I found a way around it, so I am sharing my solution here for anyone interested.
* I created a virtual environment for wisdem 3.13 and python 3.10 after reviewing the issues and documentation this was supposed to fix it. However, I still got the same message as the one in this issue so I am thinking that even wisdem 3.14 or more recent python version could be ok.
* The installer could not identify gfortran compiler, and after checking for its version in the virtual environment, it didn't exist.
* I thought it was supposed to be in the m2w64 toolchain which is a prerequisite and preinstalled in the environment.
* I did: "conda install -c conda forge m2w64-toolchain" and after that the pip install e. worked and compiled rosco properly. Tried some examples, and they run, so I think it's ok now.

dzalkind, 2024-04-17:
I think the issue is that `m2w64-toolchain` is not in the environment yaml because it is Windows specific, but it should be in our installation instructions.  Thank you for the suggestions and software tips :-)