NatLabRockies/ROSCO #328: read beamdyn?

rdamiani, 2024-03-21:
It seems to me that the main `load_from_fast` does not allow to have a beamdyn only model, is that correct?

dzalkind, 2024-03-21:
I'm not sure what you mean by a beamdyn only model.  ROSCO is looking for some data from ElastoDyn, but not much: https://github.com/NREL/ROSCO/blob/3ebe819411d5fadf630287d20bf44ff572149c63/rosco/toolbox/turbine.py#L203

rdamiani, 2024-03-21:
Apologies, I was too fast hitting submit :). It seems to me that the analogous of the following is missing from turbine.py for reading beamdyn blade properties:

`        # file
        ed_file = os.path.join(fast.FAST_directory, fast.fst_vt['Fst']['EDFile'])
        fast.read_ElastoDyn(ed_file)
        ed_blade_file = os.path.join(os.path.dirname(ed_file), fast.fst_vt['ElastoDyn']['BldFile1'])
        fast.read_ElastoDynBlade(ed_blade_file)`

I think a check on whether a ED blade file is used vs. BD blade file is used should be done and then select the correct blade file, or am i wrong?

dzalkind, 2024-03-21:
Yeah, that's a reasonable suggestion.  The only thing that ROSCO wants from an ED blade is the [damping](https://github.com/NREL/ROSCO/blob/3ebe819411d5fadf630287d20bf44ff572149c63/rosco/toolbox/turbine.py#L591), which is only used to tune the trailing edge flap control.

If you have an example and a fix, I'd be happy to help merge it in.  Otherwise, I'll leave this open until we get to it.

rdamiani, 2024-03-22:
Hi Dan, sorry for the delay, I have modified it like this, but I am still testing it out, as I am getting some issues with the read of AD data now, but I will let you know how it goes.

      ` if fast.fst_vt['Fst']['CompElast'] ==1:
            ed_file = os.path.join(fast.FAST_directory, fast.fst_vt['Fst']['EDFile'])
            fast.read_ElastoDyn(ed_file)
            ed_blade_file = os.path.join(os.path.dirname(ed_file), fast.fst_vt['ElastoDyn']['BldFile1'])
            fast.read_ElastoDynBlade(ed_blade_file)
        elif fast.fst_vt['Fst']['CompElast'] ==2:
            bd_file = os.path.join(fast.FAST_directory, fast.fst_vt['Fst']['BDBldFile(1)'])
            fast.read_BeamDyn(bd_file)
            bd_blade_file = os.path.join(os.path.dirname(bd_file), fast.fst_vt['BeamDyn']['BldFile'])
            fast.read_BeamDynBlade(bd_blade_file)
        else:
            Warning('No ElastoDyn or BeamDyn files were provided')`

dzalkind, 2024-03-22:
This is helpful...thanks!

rdamiani, 2024-03-23:
A couple of other questions (for fixed pitch cases); it seems as if ROSCO wants to change pitch even if `IPC_ControlMode=PC_ControlMode=0` ; I resorted to  setting `min_pitch= max_pitch=0`, however there are two issues with that: 
1.  controller.py checks `if not isinstance(self.min_pitch, float)` and would not take the set values if they are not specified as `0.` in the yaml, which is a bit annoying
2. ROSCO checks min '>=' max and  bombs, i think this could work with '>' which is what I did for my version, but perhaps there are other reason why you want the check to be more stringent

abhineet-gupta, 2024-07-23:
Issue with reading beamdyn resolved in #349. I'll create another issue for the question with fixed pitch case after some testing.

closed 2024-07-23 by abhineet-gupta