NatLabRockies/ROSCO #333: Output timestep in wfc_controller

HPtharaldsen, 2024-04-11:
Hi,

I am currently working with ROSCO in vscode, using Example 17b with zeromq as a baseline for my code.

 I am using wfc_controller to receive and process measurements from the simulation. Even though I've set DT and DT_out = 0.1, 0.01 etc. in my .fst-file, the measurement outputs are in a timestep of 1 (Time: 0, Time: 1, Time 2, Time: 3, Time: ..., ...) in the wfc_controller-function. Is there a way of changing the measurement ouput timestep to for example 0.1?

Thank you for any help.

Hans

dzalkind, 2024-04-11:
I would not change the DT of your OpenFAST input files.  The `ZMQ_UpdatePeriod` in the DISCON is the parameter you should change.

HPtharaldsen, 2024-04-11:
Thank you for answering. 

Even though I've updated the `ZMQ_UpdatePeriod` in the DISCON.IN-file, the timestep appears to be the same, and I haven't found any code that seems to overwrite this value. Are there any additional step i need to take after updating this parameter?

HPtharaldsen, 2024-04-12:
I solved the problem by defining the parameter in def sim_openfast():

`def sim_openfast(self):
        # Create an instance of the FAST simulation with ROSCO controller
        r = run_FAST_ROSCO()
        r.tuning_yaml = "IEA15MW_FOCAL.yaml"
        r.wind_case_fcn = cl.power_curve
        r.wind_case_opts = {
            "U": [8],
            "TMax": 1000,
        }
        run_dir = os.path.join(self.outputs, "FOCAL")
        r.controller_params = {}
        r.controller_params["LoggingLevel"] = 0
        r.controller_params["DISCON"] = {}
        r.controller_params["DISCON"]["ZMQ_Mode"] = 1
        r.controller_params["DISCON"]["ZMQ_ID"] = 1
        r.controller_params["DISCON"]["ZMQ_UpdatePeriod"] = 0.1
        r.save_dir = run_dir
        r.run_FAST()`

HPtharaldsen, 2024-04-12:
I solved the problem by defining the parameter in def sim_openfast():

```
def sim_openfast(self):
        # Create an instance of the FAST simulation with ROSCO controller
        r = run_FAST_ROSCO()
        r.tuning_yaml = "IEA15MW_FOCAL.yaml"
        r.wind_case_fcn = cl.power_curve
        r.wind_case_opts = {
            "U": [8],
            "TMax": 1000,
        }
        run_dir = os.path.join(self.outputs, "FOCAL")
        r.controller_params = {}
        r.controller_params["LoggingLevel"] = 0
        r.controller_params["DISCON"] = {}
        r.controller_params["DISCON"]["ZMQ_Mode"] = 1
        r.controller_params["DISCON"]["ZMQ_ID"] = 1
        r.controller_params["DISCON"]["ZMQ_UpdatePeriod"] = 0.1
        r.save_dir = run_dir
        r.run_FAST()
```

closed 2024-04-12 by HPtharaldsen