NatLabRockies/ROSCO #340: The acquisition of PI parameters in torque control

Tomedssa, 2024-04-27:
Hello everyone，

ROSCO uses PI control to control the torque progression of wind turbines. I would like to know how this PI parameter is determined and whether it is obtained by linearizing the wind turbine near a certain operating point (such as 12m/s wind speed or other factors).

Thanks.

abhineet-gupta, 2024-04-28:
ROSCO comes with a python based [toolbox](https://github.com/NREL/ROSCO/tree/main/rosco/toolbox) that allows for tuning of these parameters for a given turbine. The tuning process can be best understood by looking at the toolbox [documentation](https://rosco.readthedocs.io/en/latest/source/rosco_toolbox.html) and the included [Example-3](https://github.com/NREL/ROSCO/blob/main/Examples/03_tune_controller.py)

dzalkind, 2024-04-29:
More information about the PI control tuning can be found in the ROSCO paper, section 3.1.2: https://wes.copernicus.org/articles/7/53/2022/

In short, yes, the turbine is linearized at the rated wind speed.

closed 2024-05-07 by dzalkind