NatLabRockies/ROSCO #341: Missing 32-bit Windows Libraries

dzalkind, 2024-04-30:
Bladed uses 32-bit windows and requires setting up MinGW and manually adding the compilers to the environment path.  I was last having issues with adding to the path without admin rights.

closed 2024-10-18 by dzalkind

hanhanhm0, 2025-01-09:
Hello, 
 I try to build a 32bit windows version dll with the latest version of ROSCO, because I want to use ROSCO controller with Bladed.

I followed the instructions in ROSCO Documentation as bellow.

_conda install m2w64-toolchain libpython
conda install cmake make
# Clone ROSCO
git clone https://github.com/NREL/ROSCO.git

# Compile ROSCO
cd ROSCO/rosco/controller
mkdir build
cd build
cmake ..                        # Mac/linux only
cmake .. -G "MinGW Makefiles"   # Windows only
make install_

The final DLL obtained is still a  64-bit version, because I encountered the following error message.

Windows Error 193: either the dll is not a valid Win32 application, or a 32-bit process is trying to call a 64-bit process when calling 'libdiscon.dll'.

Could you tell me how to solve this problem? It will be very helpful for me. Thank you very much.

dzalkind, 2025-01-09:
Hi,

The first two steps you shared are for 64-bit windows compilers.  You'll need to install MinGW for 32-bit Windows and ensure the compilers are findable by cmake.

Best, Dan

hanhanhm0, 2025-01-14:
I have successfully compiled 32 bit dll according to your instructions. Thanks a lot.

Share my operation with friends who have the same needs as me.

I installed MinGW for 32-bit Windows, following https://capsis.cirad.fr/capsis/documentation/mingw-installation.
Delete mingw-w64 which I have already installed.
_cd ROSCO/rosco/controller
mkdir build
cd build
cmake .. -G "MinGW Makefiles" # Windows only
make install_