NatLabRockies/ROSCO #370: An issue with loading the simulator and controller interface in ROSCO

Nora-Hosseini, 2024-07-23:
Hello,

I am running ROSCO controller for 5MW NREL. However, I'm encountering an OSError in my Python script while running a simulation using ROSCO Toolbox. The specific error message is "access violation reading 0x0000000000000008". This occurs on the last step of running ROSCO, after generating DISCON.IN, when calling a function in control_interface.py, which in turn invokes a DISCON function. The traceback suggests a problem with memory access or pointers, possibly related to how data is passed to or from the DISCON function.

Does this show that there is an issue with ROSCO Toolbox or Python libraries that interface with C/C++ code? Any suggestions on how to troubleshoot or fix this OSError would be greatly appreciated.
[image]

dzalkind, 2024-07-24:
Hi @Nora-Hosseini,

This looks like a segmentation fault in ROSCO or an issue with the c-binding, which can be difficult to debug.  The troubleshooting procedure can be tedious to track these down.  

Are you running example 04_simple_sim.py on the `main` branch?  On a mac or linux?

My first step would be to add a `PRINT, * "RUNNING ROSCO"` statement around [here](https://github.com/NREL/ROSCO/blob/d6219569302fd4d8f2b8a3a0e4e72f6e9d10160e/rosco/controller/src/DISCON.F90#L66) and recompile to see if it's calling the DISCON at all, or if there's an issue with the c-binding.

Without being able to reproduce your error, it'll be hard for me to participate in the debugging.  I'll tinker on our linux set up in the meantime.

Nora-Hosseini, 2024-08-05:
> Hi @Nora-Hosseini,
>
> This looks like a segmentation fault in ROSCO or an issue with the c-binding, which can be difficult to debug. The troubleshooting procedure can be tedious to track these down. [...]

Thank you for your response. I am running ROSCO for the first time, so I am trying to run ROSCO_walkthrough.ipynb on windows. I have added PRINT, * "RUNNING ROSCO" statement, but the same error showed up. Could there be a problem with compiling DISCON.f90? 
I,also, have tried the process again following instructions in Complete ROSCO Installation , but it seems like there is still a problem with loading libdiscon, which is related with 32-bit windows application error.
[image]

dzalkind, 2024-08-05:
Hi, do you need to use this particular feature of ROSCO that let's you call the dynamic library from a python routine?  Normally, ROSCO is called from an aeroelastic simulator.   I'm not sure if we are able to support this feature on 32-bit Windows machines, but we do have libraries for running Bladed.  What is your desired use case for ROSCO?

Nora-Hosseini, 2024-08-05:
> Hi, do you need to use this particular feature of ROSCO that let's you call the dynamic library from a python routine? Normally, ROSCO is called from an aeroelastic simulator. I'm not sure if we are [...]

At the moment, I am not using any particular feature of ROSCO . I am using ROSCO_walkthrough.ipynb for running 5MW NREL turbine in a 64-bit windows machine, however I am encountered with this error when loading dll file is required.

dzalkind, 2024-08-05:
Okay, I would say that this is not a particularly well-exercised part of the ROSCO repository, so you may be safe to skip this part of your explorations.  There does seem to be a related issue here: https://github.com/NREL/ROSCO/issues/350

Nora-Hosseini, 2024-08-05:
> Okay, I would say that this is not a particularly well-exercised part of the ROSCO repository, so you may be safe to skip this part of your explorations. There does seem to be a related issue here: #350

Thank you for your assistance. I have added the mentioned codes in the link you have shared, but the same error exists with loading dll library. Could it be because the library or its inputs are compatible with 32-bit windows?

dzalkind, 2024-08-05:
No, our libraries and software take quite a bit of effort to _be_ compatible with 32-bit windows.

Have you compiled ROSCO yourself during the installation or are you using pre-existing libraries?  If you compiled yourself, it should be 64-bit.  You should use a 64-bit library if you are using a 64-bit computer to call the library.  The only reason a 32-bit library would be used is if a 32-bit software (like Bladed) calls ROSCO.

I suspect there's an issue with the `_dlopen` function on Windows, or the rosco library is not locatable on the Windows PATH.

closed 2024-10-18 by dzalkind