NatLabRockies/ROSCO #374: cannot find the ROSCO input file for IEA-15-240-RWT

Fede-274, 2024-08-19:
## Description
I cannot find the file "libdiscon.dll" and "DISCON.IN" that are requested to use the ROSCO controller alongside Fast. Therefore I cannot complete the simulation with Fast. Can someone send me these files or tell me where to find them?

abhineet-gupta, 2024-08-19:
Hi @Fede-274,
Can you explain what you're trying to do in more details ? Are you trying to run a specific example in the ROSCO repository or do you have a separate turbine model (what's the source) that you are trying to get a ROSCO controller for?

Fede-274, 2024-08-19:
Hi,
I apologize in advance because I am new to Fast and so my doubts may be silly and trivial.
I downloaded the files of the IEA 15 MW wind turbine for OpenFast from github (https://github.com/IEAWindTask37/IEA-15-240-RWT).
I want to run a test case of the 15 MW just to see the outputs (for example "IEA-15-240-RWT-Monopile") and, for what I have read, I need to install ROSCO controller to in order to complete the simulation (in particular, the file "discon.dll" and "discon.in".
Unfourtanetely, I cannot find these files. I looked in the most recent release of ROSCO but I did not find anything.

Thank you in advance

abhineet-gupta, 2024-08-19:
The `DISCON.IN` file is already provided with in the IEA-15 MW repository and pointed to on line 78 here:
https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/2575756896ab45575ab1c094da76fde65a3d1d47/OpenFAST/IEA-15-240-RWT-UMaineSemi/IEA-15-240-RWT-UMaineSemi_ServoDyn.dat#L78
So you don't need to worry about DISCON.iN.

However, you need to get discon.dll yourself.
If you are familiar with anaconda then the easiest way to get discon.dll is described here:
https://rosco.readthedocs.io/en/latest/source/install.html#anaconda-download

After this you can point to the dll by changing line 77 in the following file:
https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/2575756896ab45575ab1c094da76fde65a3d1d47/OpenFAST/IEA-15-240-RWT-UMaineSemi/IEA-15-240-RWT-UMaineSemi_ServoDyn.dat#L77

Let me know if you run into any issue with these instructions.

closed 2024-08-26 by abhineet-gupta

Fede-274, 2024-08-27:
@abhineet-gupta thank you for your indication, I did the procedure using anaconda that you have described. This is the list of files that were downloaded: 
[image]
The issue is that there isn't a libdiscon.dll, what is the problem? maybe the name of the file that I need is different?
Thank you in advance