NatLabRockies/ROSCO #377: Rosco v2.7 requirements

joanaurbano6, 2024-09-09:
Hello,

I am trying to retune the ROSCO controller used in the [IEA-3.4-130-RWT](https://github.com/IEAWindTask37/IEA-3.4-130-RWT) turbine. This turbine was modeled using rosco release v2.7.0. To start, I want just to run one of the rosco 2.7. examples, to make sure that it is working correctly. I am following the docs instructions for a complete rosco installation using an anaconda virtual environment in terminal (I am using a macbook air), the only difference is that I am uploading rosco from the tag v.2.7.0 instead of the newest version. I am not being able to run any rosco example due to compatibility issues in the packages. I am no expert in these types of softwares, but when I get an error, for example: ImportError: cannot import name 'validate_with_defaults' from 'wisdem.inputs', I try to find the error online and upgrade/downgrade the package. When I am able to solve an error, I get a different one, from a different package. I could not find a list with the necessary package versions to use rosco v.2.7. With this, I would like to ask if it is even possible/advisable to use rosco 2.7. If not, should I try to use the newest version of rosco in the IEA-3.4-130-RWT turbine. Would that be a problem, given that the turbine was modeled using rosco 2.7?

Thank you very much for your time!

dzalkind, 2024-09-09:
Hi,

It's hard to maintain previous versions of ROSCO because dependencies are always changing, so I always recommend using the latest version of ROSCO.  If you must use v2.7, I think you'll need to use an older version of WISDEM in your conda environment. 

There is a [script to update DISCON files here](https://github.com/NREL/ROSCO/blob/main/Examples/19_update_discon_version.py).  

Please try this input for the IEA-3.4: [IEA-3.4-130-RWT_DISCON_v294.IN.txt](https://github.com/user-attachments/files/16936561/IEA-3.4-130-RWT_DISCON_v294.IN.txt)

Note: I changed the normal `.IN` extension to work on GitHub.

Best, Dan

joanaurbano6, 2024-09-21:
Thank you very much for your help!

closed 2024-09-21 by joanaurbano6