NatLabRockies/ROSCO #378: Error in running 02_ccblade.py

GBA217, 2024-09-10:
I'm trying to tune the controller for a 80m diameter 2MW turbine and when I run 02_ccblade, I get this error:

(rosco-envir) c:\Users\xxxx\ROSCO\Examples>python 02_ccblade.py
Using ofTools in rosco.toolbox...
-----------------------------------------------------------------------------
Loading wind turbine data for NREL's ROSCO tuning and simulation processeses
-----------------------------------------------------------------------------
Loading FAST model: Turb10ms.fst
Loading rotor performace data from text file: c:\\NREL-2p8-127_Cp_Ct_Cq.txt
Loading rotor performance data from CC-Blade.
Running CCBlade aerodynamic analysis, this may take a minute...
UserWarning: C:\Users\xxxx\AppData\Local\anaconda3\envs\rosco-envir\lib\site-packages\wisdem\ccblade\ccblade.py:970
error.  check input values.NaNs at 0/41: 1.5707963267948966 0.0 3.141591653589793
NaNs at 0/41: 1.5707963267948966 0.0 3.141591653589793
NaNs at 0/41: 1.5707963267948966 0.0 3.141591653589793
CCBlade aerodynamic analysis run successfully.
Writing rotor performance text file: c:\Users\xxxx\ROSCO\Examples\examples_out\02_Cp_Ct_Cq.Ex03.txt

I have a few questions regarding this.
1. If I have not generated a cpctcq curve before for the model, what do I give as input for "rotor_performance_filename" in the .yaml file? I provided the path to NREL-2p8-127_Cp_Ct_Cq.txt. Will that work? Or can I leave that input blank?
2. Is aerofoil corrdinates file mandatory when I'm tuning th econtroller for the first time? It is mentioned that Numcoords can be set to zero if coordinates not included. Does that mean that it is not necessary for the simulation?
3. What could cause the above mentioned error?

.yaml file that I used:
path_params: {FAST_InputFile: Turb10ms.fst, FAST_directory: c:\01_OpenFAST_MM82, rotor_performance_filename: c:\01_OpenFAST_MM82\02_Cp_Ct_Cq.Ex03.txt}
controller_params:
    LoggingLevel: 2
    F_LPFType: 1
    F_NotchType: 0
    IPC_ControlMode: 1
    VS_ControlMode: 3
    PC_ControlMode: 1
    Y_ControlMode: 0
    SS_Mode: 1
    WE_Mode: 2
    PS_Mode: 1
    SD_Mode: 0
    Fl_Mode: 0
    Flp_Mode: 0
    zeta_pc: [2.0062413749856227]
    omega_pc: [0.13653659182284006]
    zeta_vs: 0.465
    omega_vs: 0.11
    twr_freq: 0.458
    ptfm_freq: 0.2
    ps_percent: 0.8
    sd_maxpit: 0.4363
    DISCON: {PC_MinPit: -0.1745329, PC_FinePit: 0.0, DLL_FileName: libdiscon, DLL_InFile: discon, DLL_ProcName: DISCON}
    TD_Mode: 0
    PwC_Mode: 0
    ZMQ_Mode: 0
    ZMQ_UpdatePeriod: 2
    PA_Mode: 0
    Ext_Mode: 0
    U_pc: [12]
    interp_type: sigma
    max_pitch: 1.57
    min_pitch: 0.0
    vs_minspd: 0.0
    ss_vsgain: 1.0
    ss_pcgain: 0.001
    flp_maxpit: 0.1745
    WS_GS_n: 60
    PC_GS_n: 30
    tune_Fl: true
    max_torque_factor: 1.2
    IPC_Kp1p: 0.0
    IPC_Kp2p: 0.0
    IPC_Ki1p: 5.2e-9
    IPC_Ki2p: 0.0
    IPC_Vramp: [0.0, 0.0]
    filter_params: {f_we_cornerfreq: 0.20944, f_fl_highpassfreq: 0.01042, f_ss_cornerfreq: 0.6283, f_yawerr: 0.17952, f_sd_cornerfreq: 0.41888}
    open_loop: {flag: false, filename: unused, OL_Ind_Breakpoint: 1, OL_Ind_BldPitch: 0, OL_Ind_GenTq: 0, OL_Ind_YawRate: 0}
    PA_CornerFreq: 3.14
    PA_Damping: 0.707
    linmodel_tuning:
        type: none
        linfile_path: none
        lintune_outpath: lintune_outfiles
        load_parallel: false
        stability_margin: 0.1
        omega_pc: {}
    flag: true
    tuning_yaml: /home/dzalkind/Tools/ROSCO/Tune_Cases/NREL2p8.yaml
    n_pitch: 30
    n_tsr: 30
    n_U: 1
    omega_flp: 0.0
    zeta_flp: 0.0
turbine_params: {rotor_inertia: 5375828.000, rated_rotor_speed: 188.14, v_min: 3.0, v_rated: 8.4, v_max: 25.0, max_pitch_rate: 0.1745, max_torque_rate: 11000.0, rated_power: 2000000.0, bld_edgewise_freq: 1.663, bld_flapwise_freq: 0.905}

Thanks in advance!

dzalkind, 2024-09-11:
1. Any input for the performance_filename should work. This example will generate a new one.
2. I don't think the airfoil coordinates are necessary.
3. I'm not sure what causes this error.  I will ask the WISDEM team.

ptrbortolotti, 2024-09-11:
hello, the error is most likely caused by some issue with the input files. can you maybe share your setup?

GBA217, 2024-09-17:
Hello @ptrbortolotti,
Yes, I figured that the issue is coming from the brent function calculation as the residual BEM solver residual is failing to reach zero in the iterative solver (please correct me if I'm wrong). Below is the setup:

Elastodyn:
------- ELASTODYN v1.03.* INPUT FILE -------------------------------------------
xxxx; GBA; 2024-02-29 
---------------------- SIMULATION CONTROL --------------------------------------
False                  Echo        - Echo input data to "<RootName>.ech" (flag)
3                      Method      - Integration method: {1: RK4, 2: AB4, or 3: ABM4} (-)
"default"              DT          Integration time step (s)
---------------------- DEGREES OF FREEDOM --------------------------------------
True                   FlapDOF1    - First flapwise blade mode DOF (flag)
True                   FlapDOF2    - Second flapwise blade mode DOF (flag)
True                   EdgeDOF     - First edgewise blade mode DOF (flag)
False                  TeetDOF     - Rotor-teeter DOF (flag) [unused for 3 blades]
False                  DrTrDOF     - Drivetrain rotational-flexibility DOF (flag)
True                   GenDOF      - Generator DOF (flag)
False                  YawDOF      - Yaw DOF (flag)
False                  TwFADOF1    - First fore-aft tower bending-mode DOF (flag)
False                  TwFADOF2    - Second fore-aft tower bending-mode DOF (flag)
False                  TwSSDOF1    - First side-to-side tower bending-mode DOF (flag)
False                  TwSSDOF2    - Second side-to-side tower bending-mode DOF (flag)
False                   PtfmSgDOF   - Platform horizontal surge translation DOF (flag)
False                   PtfmSwDOF   - Platform horizontal sway translation DOF (flag)
False                   PtfmHvDOF   - Platform vertical heave translation DOF (flag)
False                   PtfmRDOF    - Platform roll tilt rotation DOF (flag)
False                   PtfmPDOF    - Platform pitch tilt rotation DOF (flag)
False                   PtfmYDOF    - Platform yaw rotation DOF (flag)
---------------------- INITIAL CONDITIONS --------------------------------------
0.0                    OoPDefl     - Initial out-of-plane blade-tip displacement (meters)
0.0                    IPDefl      - Initial in-plane blade-tip deflection (meters)
0.0                    BlPitch(1)  - Blade 1 initial pitch (degrees)
0.0                    BlPitch(2)  - Blade 2 initial pitch (degrees)
0.0                    BlPitch(3)  - Blade 3 initial pitch (degrees) [unused for 2 blades]
0.0                    TeetDefl    - Initial or fixed teeter angle (degrees) [unused for 3 blades]
0.0                    Azimuth     - Initial azimuth angle for blade 1 (degrees)
7.223281620282807        RotSpeed    - Initial or fixed rotor speed (rpm)
0.0                    NacYaw      - Initial or fixed nacelle-yaw angle (degrees)
0.0                    TTDspFA     - Initial fore-aft tower-top displacement (meters)
0.0                    TTDspSS     - Initial side-to-side tower-top displacement (meters)
0.0                    PtfmSurge   - Initial or fixed horizontal surge translational displacement of platform (meters)
0.0                    PtfmSway    - Initial or fixed horizontal sway translational displacement of platform (meters)
0.0                    PtfmHeave   - Initial or fixed vertical heave translational displacement of platform (meters)
0.0                    PtfmRoll    - Initial or fixed roll tilt rotational displacement of platform (degrees)
0.0                    PtfmPitch   - Initial or fixed pitch tilt rotational displacement of platform (degrees)
0.0                    PtfmYaw     - Initial or fixed yaw rotational displacement of platform (degrees)
---------------------- TURBINE CONFIGURATION -----------------------------------
3                      NumBl       - Number of blades (-)
41.1                 TipRad      - The distance from the rotor apex to the blade tip (meters)
1.1                   HubRad      - The distance from the rotor apex to the blade root (meters)
-3.5                   PreCone(1)  - Blade 1 cone angle (degrees)
-3.5                   PreCone(2)  - Blade 2 cone angle (degrees)
-3.5                   PreCone(3)  - Blade 3 cone angle (degrees) [unused for 2 blades]
0.1                    HubCM       - Distance from rotor apex to hub mass [positive downwind] (meters)
0.0                    UndSling    - Undersling length [distance from teeter pin to the rotor apex] (meters) [unused for 3 blades]
0.0                    Delta3      - Delta-3 angle for teetering rotors (degrees) [unused for 3 blades]
0.0                    AzimB1Up    - Azimuth value to use for I/O when blade 1 points up (degrees)
-3.2                   OverHang    - Distance from yaw axis to rotor apex [3 blades] or teeter pin [2 blades] (meters)
1.1                    ShftGagL    - Distance from rotor apex [3 blades] or teeter pin [2 blades] to shaft strain gages [positive for upwind rotors] (meters)
-5.0                   ShftTilt    - Rotor shaft tilt angle (degrees)
1.25                   NacCMxn     - Downwind distance from the tower-top to the nacelle CM (meters)
0.0                    NacCMyn     - Lateral  distance from the tower-top to the nacelle CM (meters)
2.45                   NacCMzn     - Vertical distance from the tower-top to the nacelle CM (meters)
0.0                    NcIMUxn     - Downwind distance from the tower-top to the nacelle IMU (meters)
[... cut ...]
0                   ! CC_GroupIndex     - First index for cable control group, should correspond to deltaL
20.000000           ! CC_ActTau         - Time constant for line actuator [s]

!------- Structural Controllers ---------------------------------------------------------
1                   ! StC_Group_N       - Number of cable control groups
0                   ! StC_GroupIndex    - First index for structural control group, options specified in ServoDyn summary output

It is mostly derived from NREL2p8MW turbine setup as I'm trying to modify it to a smaller rotor with 2MW rated power.

ptrbortolotti, 2024-09-18:
Hello, I'm sorry but I don't really understand what's going on from this text. I talked to @dzalkind and our recommendation is to go back to a working example and start changing things one by one. It looks like you've changed the airfoil polars. Maybe start there, as things could easily go wrong in those tables. Please let us know if and when you find the issue

GBA217, 2024-10-01:
Hello @ptrbortolotti, the issue was indeed with the polars list. The order was messed up in the aerodyn input file. Thanks

closed 2024-10-01 by GBA217