NatLabRockies/ROSCO #380: ROSCO Interpretation of  FA and SS tower top acceleration signals

Sam-Ramsahoye, 2024-09-30:
Hello,

I think I may have found a bug in the interpretation of the FA and SS tower top acceleration signals received by the controller.

In https://github.com/OpenFAST/openfast/blob/main/modules/servodyn/src/BladedInterface.f90 line 1009 and 1010, elements 53 and 54 are set as YawBrTAxp and YawBrTAyp respectively.

According to https://openfast.readthedocs.io/en/main/_downloads/d8bd014121d6505cb25cf49bee5eaa80/Old_FAST6_UsersGuide.pdf, these accelerations are defined in the Tower-Top/Base-Plate Coordinate System and move with the platform but don't yaw with the nacelle.

I'm not sure what the intention is here as I'm not overly familiar with the DISCON interface but ROSCO seems to assume the FA signal rotates with the nacelle, at least for the damping of tower FA dynamics. I found this after obtaining different controller behaviour running simulations at 90 degrees to my original tests.

Best,
Sam

dzalkind, 2024-09-30:
Hi Sam,

I believe you are correct about this.  It's been noted in #254, but we haven't implemented a fix yet.  I believe we can fix it by rotating the two acceleration signals by the nacelle yaw angle. 

Another workaround is to use Fl_Mode of 2.  In that case, ROSCO will use the rotational acceleration signals that do rotate with the nacelle. 

I'll close the other issue because it's better described here, but we'll leave it open until there's a good fix.

Best, Dan

dzalkind, 2025-08-01:
This has (finally!) been addressed and fixed in #456

closed 2025-08-01 by dzalkind