NatLabRockies/ROSCO #385: How to generate a Steady Wind Input file with a step of 1 m/s for the IEA 15 MW monopile?

Fede-274, 2024-10-11:
I am testing the "IEA 15 MW monopile" under steady wind (e.g. 7 m/s) just in the x direction... Now, I want to test it under a wind profile that has a step in wind velocity (e.g. 7 m/s --> 8 m/s in 1 second). I want to impose this step once that the oscillation of the tower has become stationary and analyze the new transient. How can I resolve this issue?

dzalkind, 2024-10-11:
I use these functions to do what you suggest:
- https://github.com/NREL/ROSCO/blob/632c502db8df41311554ac2f24c08a0ea52b84d9/rosco/toolbox/ofTools/case_gen/CaseLibrary.py#L158
- https://github.com/NREL/ROSCO/blob/main/rosco/toolbox/ofTools/case_gen/run_FAST.py

You can also configure the InflowWind setup in OpenFAST manually.

Fede-274, 2024-10-13:
How would I have to manually change the InflowWind setup to impose the velocity step?
The file that you are referring is the one in the folder "IEA-15-240-RWT"? (attached photo)
[image]

closed 2024-10-18 by dzalkind