NatLabRockies/ROSCO #388: Issue in running Example 12_tune_ipc.py

YongxiangLei, 2024-11-01:
Hi Dan,

I am a newer to the ROSCO toolset, and when I was testing the examples with IPC control, I found a issue presented as follows.

Could you give me some ideas on how to fix this bug? It is caused by the mpi4py? Actually I've already tested previous cases successsfully, while example 12 for IPC failed.  I checked the core number of my PC is 1, and I changed this feature r.n_cores = 4 with r.n_cores=1, still failed. 

[image]

I would be so grateful if you could help.

Thanks in advance for your precious time.

Regards,
Yong

dzalkind, 2024-11-01:
Are you able to `conda uninstall mpi4py`  and run?

How about `mpirun -n 1 python <script>`? 

This should be fixed so that users can run without MPI even if they've installed `mpi4py`.

YongxiangLei, 2024-11-02:
Thanks for your prompt response.  

1、Sure I tried, yet still failed. I tried many other solutions on this issues,  conda uninstall mpi4py, install in other ways, even reinstall the whole rosco env...

I was wondering whether we have some other parallel compuation alternatives, such as openmpi?  or a easier way to introduce the rosco package, because I think it's a litter hard for the newer to dive into it. 

2、Again, I have another question is about the controller.py and control_interface.py, it is flexiable to modify? I mean if I would like to use the user-defined controller for IPC?  

Thanks a lot

YongxiangLei, 2024-11-02:
Hi Dan,

Regarding this issue, I fixed it by using the command mpiexec -n python <script>. It is known and caused by the conflict of system mpi and conda mpi. 

Regarding the second issue, do you have some easiest suggestions on how to interact with the openfast and rosco? I mean to extract states from the arrSways, and then output the control command for IPC. More specifically, If I have an external reinforcement learning (such as PPO) controller, I need the states and then give the action (3 IPC commands), which part I should modify? May I directly change the class in control_interface.py and controller.py? Thanks

dzalkind, 2024-11-04:
Thanks for letting us know what worked for you.

You can see the internal variables and the avrSWAP by changing the `LoggingLevel` input of ROSCO.  It will output some `RO.dbg` files with the information you seek.

You should be able to change any of the classes in this repository.

The examples with ZeroMQ may be helpful in your application: https://github.com/NREL/ROSCO/blob/main/Examples/17b_zeromq_multi_openfast.py

Best, Dan

YongxiangLei, 2024-11-05:
Thanks Dan, it is really helpful.

Best regards,
Yong

closed 2024-11-22 by dzalkind