NatLabRockies/ROSCO #391: initial values of Generator Torque - IEA 15 MW monopile

Fede-274, 2024-11-12:
Hi, I am simulating IEA 15 MW monopile turbine with different steady wind speeds. 
Analyzing the control parameters (es 5 m/s in attached photo), such as the Generator Torque, I have noticed that it initially starts from very high initial conditions (about 10000 kN*m) and then drops very quickly and arrives to the values it should have at the wind speed under consideration ... the Generated Power has a similar behaviour. Clearly I suppose that this could affect also values of Generator Speed and Thrust for example.
[image]

This “initial value” is always there but, clearly, it mostly affects low wind speeds because it is further away from the steady Generator Torque value while it is closer for higher wind speeds.

I don't think that this phenomenon has physical nature, perhaps it is to help the simulation to converge numerically? if not, what could be the reason? is there a way to enter the correct intial conditions of generator torque and generator power?

Fede-274, 2024-11-12:
I understand that the Generator Torque is function of the Generator Speed (optimal value K*omega^2 if below rated conditions or saturated if above rated conditions) ... but the point is that I have this unexpected high values of Generator Torque even if I set in ElastoDyn.dat the correct initial conditions of Generator Speed for the specific wind speed. What could be my problem?

dzalkind, 2024-11-12:
I'm not sure I have enough information to answer that.  Is the initial torque not following the generator speed like the link I have set.  Can you show me what is happening instead?

Fede-274, 2024-11-12:
[image]
[image]

Here there are the pictures of Generator Speed and Generator Torque for steady wind speed of 5 m/s. Even though I set the right initial conditions of 5 rpm for such wind velocity, the Generator Torque starts from really high values (above 10000 kN*m) and then suddenly drops to the proper value ... and I believe this affects also the Generator Speed. 

Why is that strange behaviour of Generator Torque? Is it to make the simulation numerically converge?

dzalkind, 2024-11-13:
The generator torque control input is behaving as expected; it will change until the generator speed matches its set point (around 5 rpm).  For the IEA-15MW reference turbine, a minimum generator speed (`VS_MinOMSpd`, around 5 rpm) is enforced to avoid excitations with the tower resonant mode.  Without this saturation limit, at 5 m/s, the generator speed would only reach about 3 rpm. 

It's common practice to remove the first 100+ seconds of simulation output so that transients don't affect the results. If you must reduce the generator torque transients, you could reduce the initial `RotSpeed` to 3 rpm or decrease `VS_Rgn2K` in the DISCON, which sets the initial torque.

Fede-274, 2024-11-14:
ok thanks, I understand that the Generator Torque must change until the Generator Speed reaches its steady value (which, for this wind speed, is 5 rpm). 

However, by setting steady wind and correct initial conditions of pitch, angular velocity and displacement, I would not expect to have a transient on Generator Speed and also on Thrust and Fore-Aft motion (new photos), or is this normal behavior?
[image]
[image]

Unfortunately, I cannot discard the first 100 s of simulation because my interest is in seeing how the fore-aft oscillations decay over time, so I would like to understand a little more about the trend of the aforementioned parameters of interest. 

Regarding varying VS_Rgn2K in DISCON to reduce the initial value of the Generator Torque, in the DISCON file it says that this parameter only impacts when using VS_ControlMode = 1 or 3, whereas I am using VS_ControlMode = 2, so I don't think I can do that.

Thank you in advance.

dzalkind, 2024-11-14:
This is all normal behavior based on the controller settings.

closed 2024-11-22 by dzalkind