NatLabRockies/ROSCO #392: ROSCO Simulink Implementation

AaronTUB, 2024-11-12:
## Desired use case
I want to use the ROSCO controller with "Wind Speed Estimator" through Simulink with OpenFAST.

## Description
The Simulink block "ROSCO/Wind Speed Estimator/NaN break" contains the following code
```matlab
function y = fcn(u)

if isnan(u)
    a = 1;
end

y = u;
```
Variable `a` is unused. It seems to me the line `a = 1;` should be replaced by `u = 1;`.

### Steps to reproduce issue
Run any ROSCO controller with wind speed estimator active.

### Current behavior
The input is directly fed through to the output and NaNs are not replaced.

### Expected behavior
NaNs in the input should be replaced before being fed through to the output.

### Code versions
- ROSCO main branch

dzalkind, 2024-11-12:
Hi Aaron,

Please make this change.  The ROSCO Simulink version is not maintained and is only meant to serve as a learning tool.

Best, Dan

AaronTUB, 2024-11-12:
Hi Dan,

thanks a lot for the prompt response.
That's good to know.

Best,
Aaron

closed 2024-11-12 by AaronTUB