NatLabRockies/ROSCO #397: ROSCO not building with clang/flang

ndevelder, 2024-11-21:
## Desired use case
Would like to build ROSCO in the same environment as other software in the exawind stack. This is an issue on AMD gpu based machines that rely on Clang/ROCm. 

## Description
Flang 18.0.0 gives the following error:

```
     30    F90-S-0091-Constant expression of wrong data type (/lustre/orion/cfd162/proj-shared/ndevelder/exawind-manager-11142024/environments/exawind-11122024/rosco/rosco/controller/src/ROSCO_Types.f90: 445)
     31    F90-S-0091-Constant expression of wrong data type (/lustre/orion/cfd162/proj-shared/ndevelder/exawind-manager-11142024/environments/exawind-11122024/rosco/rosco/controller/src/ROSCO_Types.f90: 445)
     32      0 inform,   0 warnings,   2 severes, 0 fatal for rosco_types
  >> 33    make[2]: *** [CMakeFiles/discon.dir/build.make:130: CMakeFiles/discon.dir/src/ROSCO_Types.f90.o] Error 1
     34    make[2]: Leaving directory '/tmp/ndevelder/spack-stage/spack-stage-rosco-main-gengfosx7r5oyhgnjnaoekv2opdxdcya/spack-build-gengfos'
  >> 35    make[1]: *** [CMakeFiles/Makefile2:114: CMakeFiles/discon.dir/all] Error 2
     36    make[1]: Leaving directory '/tmp/ndevelder/spack-stage/spack-stage-rosco-main-gengfosx7r5oyhgnjnaoekv2opdxdcya/spack-build-gengfos'
  >> 37    make: *** [Makefile:139: all] Error 2
```

### Code versions
Main branch of ROSCO

### Note
I know that this isn't necessarily something the ROSCO project can support, or something that other developers have time to work on, but even if there is slow progress, would love to have this documented and someday fixed. I'll keep hacking at it too and will post if I find anything.

marchdf, 2024-11-21:
Dupe of https://github.com/NREL/ROSCO/issues/306 ;)

dzalkind, 2024-12-06:
I'm going to close this issue and reference it in #306.  

I think this is possible and not too difficult.  @marchdf seems to have a fix.  There were a few hacks that I could help figure out long term solutions, but I don't have a good setup with flang compilers.  If either of you are able to help me with that, I could give this a better effort.

closed 2024-12-06 by dzalkind