NatLabRockies/ROSCO #402: class Turbine()——genetate_rotperf_fast():`from pCrunch import Processing`

WANG-KUI-TJU, 2024-12-11:
## Desired use case
I am modifying the blade design and using '02_ccblade.py' to calculate the blade's Cp coefficient. However, since ccblade does not account for blade deformation, I decided to use the `generate_rotperf_fast()` method from the Turbine class for calculations. But I found that this method references `Processing` from `pCrunch`. After downloading `pCrunch`, I noticed that 'Processing' is not present in its `__init__.py`
[pCrunch.git](https://github.com/NREL/pCrunch/blob/master/pCrunch/__init__.py)

## Description
[image]

### Code versions

- Python 3.10
- ROSCO 2.90
-

dzalkind, 2024-12-11:
Hi,

The method you're referencing is not tested in the normal ROSCO examples, so it may have fallen outside of our support.

It's likely you will be able to make the fixes before we can support this part of the code.  Here is a template in pCrunch that you might be able to follow to do the requisite post processing of the OpenFAST runs: https://github.com/NREL/pCrunch/blob/master/examples/post_BatchRun.py

If you develop a solution, we would be glad to support it with an example/test.  I'll keep this open until we either support this part of the `Turbine` class or remove it.

Thanks for bringing this to our attention.

Best, Dan

WANG-KUI-TJU, 2024-12-12:
Thank you for your reply!
I have checked `pCrunch` different version,modules `Processing` first came in version 0.1.0,which is not supported now. Since my recent work has been focused on solving engineering problems, I might not have time to fix this issue. I'm currently facing an engineering challenge: the wind turbine manufacturers only provide thrust curves, power curves, and some basic mass inertia parameters. I need to use these parameters (outputs) to infer a series of input parameters in order to apply these turbine parameters to the design of the floating structure. I'm currently doing this by modifying publicly available reference turbines, such as the [IEA-15MW](https://github.com/IEAWindTask37/IEA-15-240-RWT). If you have any suggestions, that would be great.Once I successfully complete the project, I will work on optimizing my workflow and improving automation efficiency. At that point, I hope I can contribute to the code.

dzalkind, 2024-12-12:
At NREL, we typically do [an inverse design using WISDEM](https://wisdem.readthedocs.io/en/master/examples/16_inverse_design/tutorial.html) to solve the challenge you shared.