NatLabRockies/ROSCO #416: ZeroMQ TorqueOffset has No Effect

Ian-Ammerman, 2025-01-17:
Hi,

I am trying to use the zeroMQ interface while running OpenFAST per example 17. I have prepared the following wfc_controller:

`def wfc_controller(id, current_time, measurements):

    # Stand in cosntant control commands
    YawOffset = 0

    if current_time > 125:
        col_pitch_command = np.deg2rad(0)
    else:
        col_pitch_command = np.deg2rad(0)

    # Assign control commands
    setpoints = {}
    setpoints["ZMQ_TorqueOffset"] = 172*current_time
    setpoints["ZMQ_YawOffset"] = YawOffset
    setpoints['ZMQ_PitOffset(1)'] = col_pitch_command
    setpoints['ZMQ_PitOffset(2)'] = col_pitch_command
    setpoints['ZMQ_PitOffset(3)'] = col_pitch_command
    
    return setpoints`

I am running into a couple of challenges regarding the torque setting. 

1). I am unable to run a constant torque simulation. The closest available is constant power, but setting VS_ControlMode to 0 keeps torque at zero. 

2). Adjusting ZMQ_TorqueOffset has no effect. When running the simulation above, I get the following plots output (using a constant power torque controller - VS_ControlMode = 1):

[image]

If I set VS_ControlMode = 0 in the .yaml file and re-run with settings as above I get the following response:

[image]

I would expect to see the generator torque either a) held at saturated region 3 torque to start or b) ramp up per the wfc_controller command. I have not touched any other settings within the .yaml file and I compiled the controller myself, ensuring that the setpoint channels were included in the .f90 file beforehand. Any help is much appreciated.

dzalkind, 2025-01-21:
Hi Ian,

This discussion may be helpful: https://github.com/NREL/ROSCO/discussions/405

We recommend using `VS_ControlMode` of 2 or 3.  Constant power or constant torque control can be set using `VS_ConstPower`.

Since you're trying something new with the examples, I recommend reviewing the existing torque controller to see if any saturation limits or logic is preventing your desired outcome: https://github.com/NREL/ROSCO/blob/975ffac26dbd968e9522e375e777993cf5d623aa/rosco/controller/src/Controllers.f90#L186

I hope this helps.

Best, Dan

Hermse, 2025-01-21:
您发给我的消息已经收到，我将尽快处理，谢谢。

closed 2025-02-28 by dzalkind