NatLabRockies/ROSCO #422: Wind speed estimator `WE_Vm` seems hard to converge

WANG-KUI-TJU, 2025-02-21:
## Description
When using ROSCO V2.90 with `WE_MODE=2`, I found that `WE_Vm` and `WE_Vt` are difficult to stabilize.I looked through the comments in the code, and they are `WE_Vm    ! Mean wind speed component `、 `WE_Vt     ! Turbulent wind speed`、 `WE_Vw        ! Estimated wind speed in WSE `、but I still don't quite understand why it's unstable. Do you know the specific algorithm they implemented?

[image]

[image]

[image]

dzalkind, 2025-02-21:
Hi,

`V_m` is the mean component, which varies very slowly.  `V_t` is the turbulence component, which varies on a faster time scale.

The important part is the total `V_w`, which does appear stable.  More details can be found in Section 5.1 here: https://wes.copernicus.org/articles/7/53/2022/

Best, Dan

WANG-KUI-TJU, 2025-02-24:
Thanks, that's really helpful

closed 2025-02-24 by WANG-KUI-TJU