NatLabRockies/ROSCO #428: Transient Pitch in 14_open_loop_control.py + olc.const_timeseries

abelvanweert123, 2025-04-01:
## Desired use case
At the moment i'm using the 14_open_loop_control example so i can test different yaw manoeuvres. In this case i'm not giving any input for the pitch or the generator torque, which was given in the initial example. 

## Description

In this case i'm not giving any input for the pitch or the generator torque, which was given in the initial example. 
My wind speed is 9 m/s so i expected to be in region 2 of the windturbine and thus get a constant pitch. But when i run my yaw cases i always seem to get a weird pitch transient that lasts for about 70s and has a weird sinusoidal behavior. 

My question in this case is whether or not this is due to wrong initial conditions or due to some issues with the DISCON.INN file 
OR
If this is due to the variable pitch control? Is the variable pitch and variable speed controller on in this example? 

I also wanted to test constant yaw cases, for this i used  olc.const_timeseries but when i used this i always got following error: 'numpy.float64' object is not iterable

My second question is why i keep getting this error even if i give a list, string or scalar

### Steps to reproduce issue 1
For the first issue with the transient Pitch 
1. Remove #olc.interp_timeseries('blade_pitch', [0,20], [0,0.0873] , 'sigma') AND
    #olc.const_timeseries('generator_torque', 19624046*.5) out of the script 

2. Run the example with  olc.sine_timeseries('nacelle_yaw', 0.0524, 60)
3. You get the following pitch 
4. The NacYaw is correct tho

[image]

### Steps to reproduce issue 2
1. Do exactly the same in step 1
2. Now run the example but with olc.const_timeseries('nacelle_yaw',0)
3. Receive error: 'numpy.float64' object is not iterable

### Current behavior
Transient Pitch behavior 

### Expected behavior
Constant Pitch behavior

### Code versions
_List versions only if relevant_
- Python 3.13.2
- rosco-toolbox Version: 2.3.0
-  OpenFAST-v4.0.2

dzalkind, 2025-04-01:
Hi,

Issue 1 looks like normal control operation and start up transients for this turbine.

Issue 2 is an odd edge case that we should probably catch.  ROSCO does not actually send a yaw position command to OpenFAST, but instead a yaw rate.  If you want to set a constant yaw angle, you should set (ElastoDyn, NacYaw) and (ServoDyn, YawNeut) to the desired constant yaw angle.  

I'll leave this open until we better handle Issue 2.

I hope this helps.

Best, Dan

abelvanweert123, 2025-04-22:
Thanks Dan,

I hope this feature will be added soon.
Is there actually a good reason to send the yawrate to OpenFAST via the DISCON.INN file instead of the yaw postion (like you do with the pitch angle)?

With kind regards,
Abel Van Weert

closed 2025-07-18 by dzalkind