NatLabRockies/ROSCO #433: Measurements from OpenFAST through ZeroMQ

john-zuluaga, 2025-04-11:
Hello,

I'm currently working on the development of a custom deep learning-based controller to regulate collective blade pitch. As the controller is written in Python, I must communicate in real-time with OpenFAST through Python, which I have found to be possible through ZeroMQ. However, the measurements available, as seen in wfc_interface.yaml, don't include some of the variables that I require for my control strategy. 

I have seen that there is an ongoing [work](https://github.com/OpenFAST/openfast/pull/2439) in the OpenFAST repository for the ZeroMQ connection, but withouth knowing the progress on that, I would like to get some information, if possible, into adding other variables for the ZeroMQ measurements myself. Issue #336 touched the same topic, and it was said then that changes to the source code would be necessary. Given that, would it be possible to get some guidance onto what changes would I expect to need to do in the source code in order to be able to increase the available measurements from ZeroMQ?

Thanks for any help beforehand.

Regards.

abhineet-gupta, 2025-04-11:
Hello @john-zuluaga,
First, please note that the ongoing work on developing ZeroMQ based connection in OpenFAST (https://github.com/OpenFAST/openfast/pull/2281) is separate form the ZeroMQ based controller interfacing capability in ROSCO. 

Second, about adding variables to the ZeroMQ measurement list ([shown here](https://github.com/NREL/ROSCO/blob/main/rosco/controller/rosco_registry/wfc_interface.yaml)), can you take a look at this [file](https://github.com/NREL/ROSCO/blob/main/rosco/controller/rosco_registry/rosco_types.yaml) and see if the variables you want to add can be found under the `LocalVariables` heading?

Best
Abhineet

john-zuluaga, 2025-04-14:
@abhineet-gupta Thanks for your answer.

Yes, the variables that I am interested in are actually included in the LocalVariables heading. Can they be made available as a ZeroMQ measurement?

abhineet-gupta, 2025-04-15:
@john-zuluaga,
All you would need to do is add the variable in [this](https://github.com/NREL/ROSCO/blob/main/rosco/controller/rosco_registry/wfc_interface.yaml) file under the `measurements` heading. Then you should be able to read that measurement like others.

Let me know in case you run into an issue with this.

john-zuluaga, 2025-04-16:
I have modified that file to add 5 additional variables and also modified the files `ZeroMQInterface.f90` and `zmq_client.c` using `write_wfc_interface.py`. However, when running an example case, I get an error at line 504 of `control_interface.py` because the `_get_measurements` function now expects to receive 22 measurements, but is only receiving the 17 original measurements defined in `wfc_interface.yaml` from the ZeroMQ server. 

I must be missing something else that has to be modified but unsure what it could be.

abhineet-gupta, 2025-04-21:
Hi @john-zuluaga 
Can you send me some variables that you are trying to add? I'll try it and let you know the solution to the error.

dzalkind, 2025-04-21:
[Running this script](https://github.com/NREL/ROSCO/blob/main/rosco/controller/rosco_registry/write_wfc_interface.py) will update `LocalVariables` to [the ZMQ interface](https://github.com/NREL/ROSCO/blob/main/rosco/controller/src/ZeroMQInterface.f90).  You'll also need to re-build ROSCO.

john-zuluaga, 2025-04-28:
Apologies for the late reply. @dzalkind 's suggestion has worked and I was able to add the variables that I wanted to the ZMQ interface. Now what if I wanted to add variables that are not included in `LocalVariables`? For example, outputs like `WindMeas1` from the lidar simulator that is included in InflowWind. Would that be possible?

Thanks to both of you for your help.

dzalkind, 2025-04-28:
You will need to find a way to add them to the `LocalVariables` in ROSCO.

[Here is where the avrSWAP is read in ROSCO](https://github.com/NREL/ROSCO/blob/cfda6af95cda15616b840199c24fbd208e63893c/rosco/controller/src/ReadSetParameters.f90#L59).  You may need to update that.

The indices of the lidar outputs in the avrSWAP can be found by printing a summary file from ServoDyn or by investigating the [extended Bladed Interface here](https://github.com/OpenFAST/openfast/blob/main/modules/servodyn/src/BladedInterface_EX.f90).

I hope this helps.

Best, Dan

john-zuluaga, 2025-04-29:
Yes, I was able to successfully add the new variables and work with them on an example case, so I will be closing this issue.

Thanks for all the help provided @abhineet-gupta and @dzalkind.

Regards,
John.

closed 2025-04-29 by john-zuluaga