NatLabRockies/ROSCO #443: NREL/ROSCO_seek for help:ERROR in ROSCOrun.m __ [Param,Cx] = ReadWrite_FAST(fast);

Crystal0912, 2025-05-14:
ROSCOrun.m
GetFASTPar:: Parameter NTypes not found.
GetFASTPar:: Parameter PerfFileName not found.
错误使用 Pre_LoadRotPerf
Error loading file

出错 ReadWrite_FAST (第 49 行)
Cx = Pre_LoadRotPerf(fullfile(discon_dir,PerfFileName(2:end-1)));

出错ROSCOrun.m (第 29 行)
[Param,Cx] = ReadWrite_FAST(fast);

Hello,
I encountered the problem above. Unfortunately, I have no idea about this after checking other issues and discussions. Could everyone please help me how solved it?

dzalkind, 2025-05-14:
Hi, 

We cannot currently test and support the Matlab_Toolbox and Simulink version of the ROSCO controller.  It should only be used as a learning tool at this time because the Simulink ROSCO controller is not up to date with the Fortran dynamic library. 
 
Since Matlab/Simulink is widely used, we have left the tools online but cannot support issues with the code.  We rely on community support to help debug issues: https://github.com/NREL/ROSCO/issues

From the error messages, it looks like the script is not finding your rotor performance file (Cp_Ct_Cq.txt).  Try with absolute paths and unicode characters.

I hope this helps.

Best, Dan

Crystal0912, 2025-05-14:
Dear @dzalkind ,
 Thank you for your reply. I have placed the relevant files (Cp_Ct_Cq. NREL5MW. txt) in the correct location, but there are also the aforementioned errors. I don't know how to handle this problem.

Camille0327, 2025-08-17:
> Hi,
>
> We cannot currently test and support the Matlab_Toolbox and Simulink version of the ROSCO controller. It should only be used as a learning tool at this time because the Simulink ROSCO [...]

I have encountered the same problem, and once I add the relevant file(Cp_Ct_Cq. NREL5MW. txt) , the problem is solved. Thanks for your help!