NatLabRockies/ROSCO #444: Usage of ROSCO dll as controller in Flex5

GBA217, 2025-05-20:
Hello,

Can any version of ROSCO controller be used with Flex5? If not, is there any modifications to the controller that will make it possible?

Thanks!

dzalkind, 2025-05-20:
Hi,

I'm not familiar with the requirements of Flex5 controller libraries.  Can you share them with me?

ROSCO compilation instructions can be found here: https://rosco.readthedocs.io/en/latest/source/install.html#compile-using-cmake

If 32-bit windows is required, the installation would be similar to the set up for Bladed: https://rosco.readthedocs.io/en/latest/source/ROSCO_instructions_for_Bladed.html

In the DISCON.IN input file, you'll want to set the `Ext_Interface` to 0.

I hope this helps.

Best, Dan

dzalkind, 2025-07-18:
Please let us know if you have any further questions.