NatLabRockies/ROSCO #445: Missing 'FAST2Matlab'

Cadulo, 2025-05-27:
Hello,

I'm trying to run the co-simulation with ROSCO using the latest version of the MATLAB toolbox as recommended [here](https://github.com/NREL/ROSCO/tree/main). However, I'm encountering a missing function error when executing the runFAST.m script:

>> runFAST

fast = 

  struct with fields:

     FAST_SFuncDir: 'C:\Open Fast 4.0.2\Matlab-Simulink'
    FAST_InputFile: 'IEA-3.4-130-RWT.fst'
    FAST_directory: 'C:\Open Fast 4.0.2\IEA-3.4-130-RWT-master\openfast'

Unrecognized function or variable 'FAST2Matlab'.

Error in ReadWrite_FAST (line 14)
FP = FAST2Matlab(fullfile(fast.FAST_directory,fast.FAST_InputFile),2); %FP are Fast Parameters, specify 2 lines of header (FAST 8)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Error in runFAST (line 36)
[Param,Cx] = ReadWrite_FAST(fast);
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

I've checked the repository and it seems like the FAST2Matlab function is missing. I followed the setup instructions carefully and used the develop branch of the [matlab-toolbox](https://github.com/NREL/matlab-toolbox) as indicated in the ROSCO documentation.

Is this repository incomplete? Or am I missing an additional step or dependency to include FAST2Matlab?

Any help would be appreciated.

Thanks in advance!

williamjussiau, 2025-06-06:
```Unrecognized function or variable``` can be raised if the function has not been found in the Matlab path, which could maybe be the case here? The matlab-toolbox files are most likely in another folder that you might want to permanently add to your Matlab path with ```addpath+savepath```?

dzalkind, 2025-07-18:
Redirecting this to the main Matlab/Simulink issue of #263

closed 2025-07-18 by dzalkind