NatLabRockies/ROSCO #446: Function or variable 'OutList' is not recognized.

sunmaxin, 2025-06-03:
When I using the 'runFAST.m' to run openfast_simulink, it appears“Incorrect use of runFAST：Function or variable 'OutList' is not recognized."and"An error occurred while evaluating the expression 'BasicModel_Lookuptable_V4/DFIG/Extract Signals/Extract GenSpeed'."
I don't know how to fix this, and I'd like to get your help. Thanks!
Best regards.

sunmaxin, 2025-06-03:
Here is my code setting.
clear;
fast.FAST_SFuncDir     = 'C:\Users\admin\Desktop\openfast a\example\1\simulink';  %%%% NEED FOR SIMULINK指定Simulink的FAST函数目录
fast.FAST_InputFile    = '5MW_Land_Simulink.fst';   % FAST input file (ext=.fst)FAST的输入文件（.fst文件）
fast.FAST_directory    = 'C:\Users\admin\Desktop\openfast a\example\1\ROSCO-main2.6\Test_Cases\5MW_Land_Simulink';   % Path to fst directory files

% Simulink Parameters
% Model
simu.SimModel           = 'C:\Users\admin\Desktop\openfast a\example\1\ROSCO-main2.6\Matlab_Toolbox\Simulink\BasicModel_Lookuptable_V4.slx';

% Script for loading parameters
simu.ParamScript        = 'C:\Users\admin\Desktop\openfast a\example\1\ROSCO-main2.6\Matlab_Toolbox\Utilities\load_ROSCO_params';

%% Simulink Setup

[ControlScriptPath,ControScript] = fileparts(simu.ParamScript);
addpath(ControlScriptPath);
addpath(fast.FAST_SFuncDir);
addpath('C:\Users\admin\Desktop\openfast a\example\1\ROSCO-main2.6\Matlab_Toolbox\Utilities')

%% Read FAST Files & Load ROSCO Params from DISCON.IN

[Param,Cx] = ReadWrite_FAST(fast);
disp('Loaded Param:');
disp(Param);
% Simulation Parameters
simu.TMax   = Param.FP.Val{contains(Param.FP.Label,'TMax')};
simu.dt   = Param.FP.Val{contains(Param.FP.Label,'DT')};
[R,F] = feval(ControScript,Param,simu);

%% Premake OutList for Simulink

%OutList = {'Time'};
%OutList = [OutList;
  %  Param.IWP.OutList;
  %  Param.EDP.OutList;
  %  Param.ADP.OutList;
  %  Param.SvDP.OutList;
  %  ];

%disp('OutList contents:');
%disp(OutList);

%for iOut = 2:length(OutList)
%   OutList{iOut} = OutList{iOut}(2:end-1); %strip "s
%end

%% Premake OutList for Simulink

OutList= {'Time'};
OutList2 = {'Time'};
OutList2 = [OutList2;
Param.IWP.OutList;
Param.EDP.OutList;
Param.ADP.OutList;
Param.SvDP.OutList;
];
%disp('OutList contents:');
%disp(OutList);
for iOut = 2:length(OutList2)
OutList = [OutList;OutList2{iOut}(1:end)]; %strip "s
end

for iOut2 = 1:length(OutList)
OutList{iOut2}=strrep(OutList{iOut2},'"','');
end
disp('OutList contents:');
disp(OutList);
disp('OutList2 contents:');
disp(OutList2);
%% Exectute FAST

% Using Simulink/S_Func
FAST_InputFileName = [fast.FAST_directory,filesep,fast.FAST_InputFile];
TMax               = simu.TMax;

SimulinkModel = simu.SimModel;

Out         = sim(SimulinkModel, 'StopTime', num2str(simu.TMax));
sigsOut     = get(Out,'sigsOut');   %internal control signals

%% Get OutData
SFuncOutStr = '.SFunc';

% Try text first, then binary
[OutData,OutList] = ReadFASTtext([fast.FAST_directory,filesep,fast.FAST_InputFile(1:end-4),SFuncOutStr,'.out']);
if isempty(OutData)
    [OutData,OutList] = ReadFASTbinary([fast.FAST_directory,filesep,fast.FAST_InputFile(1:end-4),SFuncOutStr,'.outb']);
end

% Dump data to structure
for i = 1:length(OutList)
    simout.(OutList{i}) = OutData(:,i);
end

%% Plot
Pl_FastPlots(simout)

sunmaxin, 2025-06-03:
>> runFAST
Warning: There are more headers in the table than columns. Ignoring the last 4 column(s). Problematic line:0.0000000E+00  6.0000000E+00  1.0000000E+00  1.000000E-01  
Loaded Param:
         FP: [1×1 struct]
        EDP: [1×1 struct]
        IWP: [1×1 struct]
        ADP: [1×1 struct]
       SvDP: [1×1 struct]
    SD_dllP: [1×1 struct]

OutList contents:
    {'Time'     }
    {'Wind1VelX'}
    {'Wind1VelY'}
    {'Wind1VelZ'}
    {'OoPDefl1' }
    {'IPDefl1'  }
    {'TwstDefl1'}
    {'BldPitch1'}
    {'Azimuth'  }
    {'RotSpeed' }
    {'GenSpeed' }
    {'TTDspFA'  }
    {'TTDspSS'  }
    {'TTDspTwst'}
    {'Spn2MLxb1'}
    {'Spn2MLyb1'}
    {'RootFxb1' }
    {'RootFyb1' }
    {'RootFzb1' }
    {'RootMxb1' }
    {'RootMyb1' }
    {'RootMzb1' }
    {'RotTorq'  }
    {'LSSGagMya'}
    {'LSSGagMza'}
    {'YawBrFxp' }
    {'YawBrFyp' }
    {'YawBrFzp' }
    {'YawBrMxp' }
    {'YawBrMyp' }
    {'YawBrMzp' }
    {'TwrBsFxt' }
    {'TwrBsFyt' }
    {'TwrBsFzt' }
    {'TwrBsMxt' }
    {'TwrBsMyt' }
    {'TwrBsMzt' }
    {'NcIMURAys'}
    {'NcIMUTAxs'}
    {'RtSpeed'  }
    {'GenPwr'   }
    {'GenTq'    }

OutList2 contents:
    {'Time'   }
    { 3×1 cell}
    {35×1 cell}
    { 1×1 cell}
    { 2×1 cell}

 **************************************************************************************************
 OpenFAST

 Copyright (C) 2021 National Renewable Energy Laboratory
 Copyright (C) 2021 Envision Energy USA LTD

 This program is licensed under Apache License Version 2.0 and comes with ABSOLUTELY NO WARRANTY.
 See the "LICENSE" file distributed with this software for details.
 **************************************************************************************************

 OpenFAST-v3.0.0-dirty
 Compile Info:
  - Compiler: Intel(R) Fortran Compiler 1900
  - Architecture: 64 bit
  - Precision: single
  - OpenMP: No
  - Date: Jun 23 2021
  - Time: 21:06:06
 Execution Info:
  - Date: 06/03/2025
  - Time: 09:51:43+0800

 OpenFAST input file heading:
     FAST Certification Test #18: NREL 5.0 MW Baseline Wind Turbine (Onshore)

 Running ElastoDyn.
 Nodal outputs section of ElastoDyn input file not found or improperly formatted.
 Running AeroDyn.
 AD15 Nodal Outputs: Nodal output section of AeroDyn input file not found or improperly formatted.
 Skipping nodal outputs.
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 1, blade 1)
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 2, blade 1)
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 3, blade 1)
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 4, blade 1)
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 1, blade 2)
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 2, blade 2)
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 3, blade 2)
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 4, blade 2)
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 1, blade 3)
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 2, blade 3)
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 3, blade 3)
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 4, blade 3)
 Running InflowWind.

    Reading a 31x31 grid (178 m wide, 1 m to 179 m above ground) with a characteristic wind speed
    of 12 m/s. This full-field file was generated by TurbSim (v2.00.08a-bjj, 18-Jul-2017) on
    30-May-2025 at 16:34:59.

    Processed 12600 time steps of 20-Hz full-field data (period of 630 seconds).
 Running ServoDyn.
  Time: 0 of 100 seconds.

  OpenFAST completed.
Error using runFAST
Error due to multiple causes.
Causes:
Error using runFAST
Function or variable 'OutList' is not recognized.
Error using runFAST
Error in evaluating expression 'BasicModel_Lookuptable_V4/DFIG/Extract Signals/Extract GenSpeed'.

dzalkind, 2025-07-18:
Redirecting this to the main Matlab/Simulink issue of #263

closed 2025-07-18 by dzalkind