NatLabRockies/ROSCO #448: Figure 4 of Rosco document

TheMercer, 2025-06-24:
Do you happen to have the code to produce this plot ? 

I assume it was done more automatically based of the controller settings and performance curves ?

Not hard coded ? 

If you have the code, would I be able to have it ? 

Thank you

dzalkind, 2025-06-26:
Hi,

No, we do not have this exact code anymore.  It would be a welcome addition to this example: https://github.com/NREL/ROSCO/blob/main/Examples/02_ccblade.py

The operating points of the controller should be able to be extracted like in this example: https://github.com/NREL/ROSCO/blob/main/Examples/03_tune_controller.py

These variables are what you are looking for: https://github.com/NREL/ROSCO/blob/cfda6af95cda15616b840199c24fbd208e63893c/rosco/toolbox/controller.py#L357

Best, Dan

TheMercer, 2025-06-28:
Thank you, I will try this in the week

TheMercer, 2025-06-28:
@dzalkind how would I do it with the WindIO yaml is there an easy interface between the code ?

TheMercer, 2025-06-29:
@dzalkind, I have an issue. 

Where I have PS mode set to 3. However it's not Cp maxise. PS mode 0, 1, 2, 3 all look the same. My model isn't far / basically the IEA 22MW. Do you know why. 

I checked my installed scripts of Rosco and they match the repo. 

Are you able to show the performance graph for 22MW. 

If I set my min pitch to -0.07 råd (-4deg) same as 22MW my Rosco tuning or WEIS tuning saturates to that limit which obviously isn't Cp maxing.

Is this a bug somewhere? (I have messed with numbers as much as possible and can't get it to CP max

TheMercer, 2025-06-29:
I found the issue with my current set up, probably issue in full if you can check also.

If i set min_pitch in tuning yaml to 0 its the same as the default, however if set to 0.0 it changes the results. 

This is due to in the tuning setup you check if min_pitch is a float in a few places however 0 is and int and 0.0 is a float so you get two different behaviours

TheMercer, 2025-06-29:
[image]

TheMercer, 2025-06-29:
On the right i got the legend of the graph wrong way around. 

Basically looks like if your min_pitch is int(0) it will not do peak shaving. If it float(0) them it will peak shave

TheMercer, 2025-06-30:
Also set up tuning yaml for 22MW and same issues. Also was the min pitch curve edited Manual for 22MW ? As i put with -0.07 min pitch and it Saturates  to that angle but if it was CP maxing it wouldn't saturate to the -0.07

TheMercer, 2025-07-18:
Yes exactly, it's just a weird edge case and from a general point of view, 0 should equal 0.0 in behaviour. I guess it's to get fine pitch behaviour? 

So maybe a rosco flag to intend that behaviour instead of relying onto int or float or flag it or print in red/blød etc thats the behaviour?

dzalkind, 2025-07-18:
Should be fixed in #456