NatLabRockies/ROSCO #454: Problem with yaw rate control Nacelle heading target

theot-student, 2025-07-18:
Hi everyone, I want to report a "bug" in the yaw rate control of ROSCO:

I wanted to use the yaw control from ROSCO using OpenFAST software. To test that I created a wind input that goes from 0 to 10° in the first 20 second:
[image]

My expectations were that the nacelle yaw will reach the 10°, however it never occurs but reach around 9°:
[image]

Moreover, I tried with a negative wind direction (same magnitude as before) and now it works:
[image]

So I checked the YawRateControl function in ROSCO and I found out that the NacHeadingTarget is variable scaled between 0 and 360°. Then I plot this variable and I saw it never reach -10° (150°) but converges to -9° (151°) in the first case : 
[image]

My guess was that the conversion of the WindDirection between 0 and 360° make the filter act differently for negative and positive angle (the filter does not compute the filter similarly on small and large angle values).

To check my result, I changed the wrap_360 function with a wrap_180 function in the wind filter in order to keep small negative values:
[image]

And I test again for the first wind case and now it works:
[image]

I don't know if there was a reason for this wrap_360 function or if we can really call it a bug but I think this is not an expected behaviour of the controller.

Best Regard,
T. Barbier

dzalkind, 2025-07-18:
Thanks for the notice!  I think I agree with your fix, but I've asked a few colleagues with more experience with the yaw controller to weigh in.  

I staged this fix in #455.

dzalkind, 2025-08-01:
This should be fully addressed in #456.  Both `wrap_*` functions should work but there was an issue with the `wrap_360` and a too low precision for `D2R`.

closed 2025-08-01 by dzalkind