NatLabRockies/ROSCO #462: Problems running example 24 (24_floating_feedback.py)

lucpavonese, 2025-08-13:
Hi, 

I am currently struggling to run completely the floating feedback example. I am using the ROSCO version 2.9.5 installed via pip. The script completes the generation of the DISCONs but fails to run openFAST and I believe the reason for this is that somewhere, we should define the path to the openFAST executable. So, the problem is, to run the example, where should we add the openFAST executable?

> OpenFAST Failed: [WinError 2] The system cannot find the file specified
> Runtime:        IEA15MW_1.fst = 0.00  s
> OpenFAST Failed! Please check the run logs.
> Writing new controller parameter file parameter file: c:\User\Desktop\Tuning ROSCO\examples_out\24_floating_feedback\IEA15MW\simp_step\base\IEA15MW_2_DISCON.IN.
> Writing new controller parameter file parameter file: c:\User\Desktop\Tuning ROSCO\examples_out\24_floating_feedback\IEA15MW\simp_step\base\IEA15MW_3_DISCON.IN.
> OpenFAST Failed: [WinError 2] The system cannot find the file specified
> Runtime:        IEA15MW_0.fst = 0.00  s
> OpenFAST Failed! Please check the run logs.
> OpenFAST Failed: [WinError 2] The system cannot find the file specified
> Runtime:        IEA15MW_2.fst = 0.00  s
> OpenFAST Failed! Please check the run logs.
> OpenFAST Failed: [WinError 2] The system cannot find the file specified
> Runtime:        IEA15MW_3.fst = 0.00  s
> OpenFAST Failed! Please check the run logs.
> multiprocessing.pool.RemoteTraceback: 
> """
> Traceback (most recent call last):
>   File "C:\Program Files\Python311\Lib\multiprocessing\pool.py", line 125, in worker
>     result = (True, func(*args, **kwds))
>                     ^^^^^^^^^^^^^^^^^^^
>   File "C:\Program Files\Python311\Lib\multiprocessing\pool.py", line 48, in mapstar
>     return list(map(*args))
>            ^^^^^^^^^^^^^^^^
>   File "C:\User\AppData\Python\Python311\site-packages\rosco\toolbox\ofTools\case_gen\runFAST_pywrapper.py", line 441, in evaluate_multi
>     return evaluate(indict)
>            ^^^^^^^^^^^^^^^^
>   File "C:\User\AppData\Python\Python311\site-packages\rosco\toolbox\ofTools\case_gen\runFAST_pywrapper.py", line 436, in evaluate
>     return fast.execute()
>            ^^^^^^^^^^^^^^
>   File "C:\User\AppData\Python\Python311\site-packages\rosco\toolbox\ofTools\case_gen\runFAST_pywrapper.py", line 177, in execute
>     raise Exception('OpenFAST Failed! Please check the run logs.')
> Exception: OpenFAST Failed! Please check the run logs.
> """
> 
> The above exception was the direct cause of the following exception:
> 
> Traceback (most recent call last):
>   File "c:\User\Desktop\Tuning ROSCO\24_floating_feedback.py", line 153, in <module>
>     main()
>   File "c:\User\Desktop\Tuning ROSCO\24_floating_feedback.py", line 120, in main
>     r.run_FAST()
>   File "C:\User\Python\Python311\site-packages\rosco\toolbox\ofTools\case_gen\run_FAST.py", line 200, in run_FAST
>     fastBatch.run_multi(cores=self.n_cores)
>   File "C:\User\AppData\Python\Python311\site-packages\rosco\toolbox\ofTools\case_gen\runFAST_pywrapper.py", line 342, in run_multi
>     output = pool.map(evaluate_multi, case_data_all)
>              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "C:\Program Files\Python311\Lib\multiprocessing\pool.py", line 367, in map
>     return self._map_async(func, iterable, mapstar, chunksize).get()
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "C:\Program Files\Python311\Lib\multiprocessing\pool.py", line 774, in get
>     raise self._value
> Exception: OpenFAST Failed! Please check the run logs.
> C:\Program Files\Python311\Lib\multiprocessing\pool.py:268: ResourceWarning: unclosed running multiprocessing pool <multiprocessing.pool.Pool state=RUN pool_size=4>

Cheers!

dzalkind, 2025-08-13:
Hi,

You can simplify the script and probably get a better error message by [setting `n_cores = 1`](https://github.com/NREL/ROSCO/blob/03292150c047fe2904bf1316a76a6d26cee109ac/Examples/24_floating_feedback.py#L119).

The path to the openfast executable is [an attribute of the `run_FAST_ROSCO` class](https://github.com/NREL/ROSCO/blob/03292150c047fe2904bf1316a76a6d26cee109ac/rosco/toolbox/ofTools/case_gen/run_FAST.py#L61).  You can set `r.openfast_exe` to your desired path.

Best, Dan

lucpavonese, 2025-08-14:
Perfect Dan, it is working now. Thanks!